RasPlex/RasPlex #593: [Feature Request] Screensaver to turn off HDMI

schmitzCatz, 2017-09-10:
Hi Guys,

it would be nice to have a screen saver to turn off HDMI, this would enable an attached monitor or tv to switch off. Maybe it is easy to implements with a python screen saver which uses:

```
//turn off hdmi
vcgencmd display_power 0
//turn on hdmi
vcgencmd display_power 1
```

Cheers!

schmitzCatz, 2018-01-22:
HI,

I found a brave soul who developed a python screensaver to archive this goal. Works perfecly with my rasplex client.

https://github.com/dllz/script.screensaver.xbian-hdmi-off

Thanks to dllz.

ToxicFrog, 2018-08-24:
How do you install this? Rasplex doesn't seem to have the addon management screen that stock Kodi or LibreElec do.