RasPlex/RasPlex #594: No more correct skin for live tv

Apadgett879, 2017-09-14:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: Raspberrypi model 3

**RasPlex**: Rasplex version 1.8  default skin

**PMS**:  (will provide if needed)

**Network**:  Wifi (802.11 N) 

My issue is that accourding to the link below (your how to get live TV on rasplex from server guide) you want us to switch from MediaStream to Confluence, however there seem to be no options for either of those skins in rasplex 1.8

https://github.com/RasPlex/RasPlex/wiki/LiveTV

g1t-dlanor, 2017-09-28:
That Wiki was created for a very old (== obsolete) version of RasPlex, so it's no longer valid.
Those old skins and methods are simply not compatible with current RasPlex or OpenPHT versions.

g1t-dlanor, 2017-09-28:
In fact I'm not sure if some of the methods might still work or not, but to try them out you'd need to install the old skins (possibly customized) through some file transfer method. Each custom skin folder needs to be stored inside the RasPlex system folder "/storage/.plexht/addons/". Then they will be recognized for selection in the skin choice menu of RasPlex. But you need to realize that the old Mediastream and Confluence skins were not designed for the current RasPlex program and may lock up when you try to use some of the old skin menus.