RasPlex/RasPlex #596: [Feature Request] Shairport Sync

AlekLT, 2017-10-08:
Please use [Shairport Sync](https://github.com/mikebrady/shairport-sync) for Airplay support. It's up to date, well maintained and has far better support than the existing implementation. Because RasPlex uses LibreELEC, it's not possible to install Shairport Sync after installation, and so RasPlex just can't support Airplay on current devices.

Rtaxerxes, 2018-02-10:
+1 Would very much like Airplay support again.

faithless1108, 2018-04-29:
+1 Verry usefull feature!

holysoles, 2020-05-10:
+1 I would love to see this added!

jakehopking, 2021-02-04:
+1 When I found RasPlex today, I thought it had solved my needs. However as there is no shairport-sync, I'm unfortunately going to have to look elsewhere.