RasPlex/RasPlex #600: Lets try this again... Keyboard shortcuts.

Darkassassin07, 2017-11-06:
Closed
Keyboard shortcuts #599
@Darkassassin07
Darkassassin07 opened this issue 
about 2 hours ago 
Is there somewhere i can find a list of what controls are bound to what keys?
There are keys/controls that are not expected and really annoying:

The up/down arrows will bring up the player controls about 30% of the time, the rest of the time it exits the video forcing you to restart in the case of a channel.

The backslash key changes the resolution (a keybind for it? Really?) but is placed directly above the enter key... Far to easy to press accidentally making you exit what ever you were doing and fix it in settings.

I have accidentally pressed other keys and combinations that have done wierd things like restart the vid of exit to home.

Why is this not easy to find?
@bkury
bkury 
commented about 2 hours ago
Keymap file:
https://github.com/RasPlex/OpenPHT/blob/openpht-1.8/system/keymaps/keyboard.xml

You can edit this file to suit your needs :
Windows: C:\Program Files (x86)\OpenPHT\system\keymaps\keyboard.xml
Linux: /usr/share/openpht/system/keymaps/keyboard.xml

bkury closed this issue about 2 hours ago

Darkassassin07 
commented less than a minute ago
This is not closed...

/usr/share/openpht
Does not exist on the rasplex filesystem....

Can i attempt a solution and see if it works befor you close this one?

bkury, 2017-11-06:
Sorry, I thought that you can re-open it yourself if something goes wrong ...

RasPlex: /usr/share/XBMC/system/keymaps/keyboard.xml

You might want to copy this file to "/storage/.plexht/userdata/keymaps/keyboard.xml" (if it's not already there) to be able to modify it ...

Wiki: https://github.com/RasPlex/RasPlex/wiki/Remote-Controls

Darkassassin07, 2017-11-07:
Thanks @bkury found the file :)

closed 2017-11-07 by Darkassassin07