RasPlex/RasPlex #601: Bluetooth audio disconnects

bullj, 2017-11-12:
I am able to connect to my Samsung Wireless Audio with Dock DA-E651 for audio, but after several minutes of playback, or after selecting a new video to play, audio output stops with no error message to screen.  
Logs available at: http://sprunge.us/HRGR

Thanks in advance.

**RPi model**: Raspberry Pi 3 Model B
**RasPlex**: 1.8.0 default skin
**PMS**: 1.9.7.4441 on Linux Mint
**Network**: ethernet

WireCrazii, 2018-01-02:
Same problem.

Found that if you use internal bluetooth of rPi audio will work only first few seconds. But if you use it with usb bluetooth dongle works without problem.

**RPi model**: Raspberry Pi 3 Model B
**RasPlex**: 1.8.0 default skin
**PMS**: 1.10.1.4602 on Windows
**Network**: ethernet

selfsimilar, 2018-03-17:
Same issue, I assume. Connecting to AmazonBasics Bluetooth 4.0 Audio Receiver is fine, but playing video will reboot client.

**RPi model:** Raspberry Pi Zero W
**RasPlex**: 1.8.0 default skin
**PMS**: 1.12.0.4829 on FreeNAS
**Network**: wireless

hgfhj658d4fg64dfg, 2018-09-14:
I am having the same problem. I can not reconnect to the bluetooth device until I delete it, reboot and re-pair.