RasPlex/RasPlex #606: Doesn't work with Raspberry Pi 3 B+

ben423423n32j14e, 2018-03-17:
Pi sits on rainbow boot screen with yellow power icon at top right.

Support needs to be added for the 3 B+

yishaib1, 2018-03-21:
+1 
RasPlex its an amazing project!

aidanb, 2018-03-21:
I am having the same issue. I installed the Pi2 version of Rasplex as suggested here and can only see the yellow power icon.

pigetto, 2018-03-22:
+1
Same issue, rainbow screen at boot.
Eager to try out my new RPi3+ with RasPlex, especially interresting to see improvements in the network speed.

ambreswe2, 2018-03-23:
+1
Same issue. Please add support for the pi3 b+, Rasplex is awesome! <3

dalehamel, 2018-03-23:
Hi folks,

Waiting to get a pi 3 b+ to test on so we can support it. I expect it'll be a simple firmware update.

jesperrasmussen, 2018-03-25:
@dalehamel I'd be more than happy to run a few tests on my 3 b+ - I tried to upgrade some of the basic firmware files to the latest versions from Pi's official repository, but with no luck. I'm guessing the change also involves headers, requiring a recompile of other parts of RasPlex - haven't yet set up a development environment for doing so, as I'm a bit confused about the build process, maybe that'll be an Easter project ;)

ambreswe2, 2018-03-26:
@dalehamel I have a b+ I can test on if you guys need any help.

scott-lamson, 2018-03-27:
+1
Same issue. I was so excited to get a pi to put Rasplex on it and didnt realise the new 3 b+ wasn't compatible. I would have bought the 3 if I had known. Hope a fix comes soon

dalehamel, 2018-03-27:
Have received a board to test on, hope to get working on it this weekend.

Thanks all for your patience

ben423423n32j14e, 2018-03-28:
Just wanted to say that despite already having a 3 B+ I bought an odroid C2 which can run Rasplex at 4k resolution with HVEC hardware support and has full gigabit Ethernet, also the optional Wifi adapter from odroid is a lot faster than the Pi 3 B+ 5ghz Wifi. I'd suggest anyone who has yet to make a purchase to look at the odroid C2.

The image for it is here:
https://github.com/RasPlex/OpenPHT/releases

Sorentoman, 2018-03-31:
Seems I'm not the only one pulling my hair out trying to get new rpi 3 B+ working. Thought everything was good when I swapped out rpi 3 for new rpi 3 B+. Plex media server came up fine, browsing library looked good but I started playing a movie and it stopped about an hour into it. Wouldn't restart. Tried another movie and it played fine until I let run and it stopped after about an hour. Tried rebuilding server with latest image (3-2018) and ran into multiple issues. Swapped the original rpi 3 back with my original sd card and all is well again. Thought that I might use the new rpi 3 B+ with Rasplex but comments above don't make me optimistic. Have I bought a new rpi that I can't use?

dalehamel, 2018-03-31:
> Have received a board to test on, hope to get working on it this weekend.

@Sorentoman as stated above, work is underway to update rasplex to include the necessary fixes to support 3 B+.

I am presently building it against the following tree https://github.com/RasPlex/OpenPHT-Embedded/tree/rasplex-2018 based on LibreELEC 8.2.4. 

I'm optimistic something will be built soon, please be patient.

dalehamel, 2018-03-31:
Update: have a build that completes, needs a bit of work to be ready for testing, and make sure the necessary firmware fixes are included, which may be in LibreELEC 8.2.5.

dalehamel, 2018-03-31:
Current build seems to boot successfully, but has stability issues. Confirmed working on a 3 B+. 

Posting for anyone who is eager, but must stress that this is KNOWN to not be stable.

More debugging needed, as it spontaneously reboots for me while sitting at main screen. Seems to get further than any attempt so far though.

I believe I know what the issues is, working on fixing it.

https://github.com/RasPlex/RasPlex/releases/download/PRE-1.8.0b%2B/RasPlex-1.8.0.0-573b6d73-RPi2.arm.img.gz

dalehamel, 2018-03-31:
Update: It appears that our latest build runs well against standard RPi 3 units, but crashes on the RPi 3 b+. We expect that this is error is at the OS or firmware level, and will probably be fixed in future versions of LibreELEC (like 8.2.5 as mentioned above). 

For a fully stable release, we will need to fix this firmware issue. It probably makes the most sense to wait for LibreELEC to fix it, so a stable release may be 1 or 2 weeks out folks.

Thanks for your patience, really looking forward to getting RasPlex running smoothly on the latest RPi hardware.

dalehamel, 2018-04-01:
Got a build done off of the latest LibreELEC.

It has some crashing issues still (outlined [here](https://github.com/RasPlex/RasPlex/releases/download/PRE-1.8.0b%2B/RasPlex-1.8.0.0-573b6d73-RPi2.arm.img.gz)), but is better than nothing for anyone with a Rpi3 B+ that's not able to boot at all.

I'd currently running my main Pi off of [this build](https://github.com/RasPlex/RasPlex/releases/download/PRE-1.8.0b%2B/RasPlex-1.8.0.0-573b6d73-RPi2.arm.img.gz) as it seems to be stable enough for general use once it has had my library completely pre-cached.

As new patches are added for LibreELEC 8.2.5 that will hopefully address this (or, another source of the crashing

ambreswe2, 2018-04-01:
Sounds good :)

Proteus262, 2018-04-02:
Fantastic, thank you!

jesperrasmussen, 2018-04-02:
@dalehamel To provide some basic feedback, I've been running the build on my 3b+ since you released it - have had no stability issues yet, have never seen the lightning icon while booting/using Rasplex (have been watching shows and movies for about 8 hours with the new release) and have experienced no restarts.

My Pi is running 24/7 with the standard, official PSU.

dalehamel, 2018-04-02:
@jesperrasmussen interesting, so it could just be a power management problem.

Can you post a link to the exact power adapter you are using? The ones I had been using for testing used to be considered "good enough", but perhaps the 3 b+ has more stringent requirements.

ambreswe2, 2018-04-02:
@dalehamel Currently running the build you posted, nothing weird so far, have only tested it quickly. Thanks!

dalehamel, 2018-04-02:
anyone who has it working fine: please post the exact specs of your power supply (haw many milliamps, what voltage, and exact model number / make if possible).

Likewise, anyone with issues please do the same. Hopefully this will expose a trend, if the 3 b+ is more power hungry than prior generations.

jesperrasmussen, 2018-04-02:
@dalehamel This is the one: https://www.raspberrypi.org/products/raspberry-pi-universal-power-supply/

The complete specs on the PSU are here at the local, Danish retailer I bought mine from, as the official Raspberry Pi page does not show them for some reason. The page is in Danish, but the specs should be readable anyway: https://dk.rs-online.com/web/p/stromforsyning-netstikshus/9098126/

Also, when they announced the 3 b+, they made a note that due to the extra consumption of power in the new version, the recommendations towards what PSU are "enough" has changed - it's mentioned here (Look for "Note that Raspberry Pi 3B+ does consume substantially more power than its predecessor. We strongly encourage you to use a high-quality 2.5A power supply, such as the official Raspberry Pi Universal Power Supply."): https://www.raspberrypi.org/blog/raspberry-pi-3-model-bplus-sale-now-35/

dalehamel, 2018-04-02:
I've ordered an official RPi power supply to verify that it fixes the issue in testing.

dalehamel, 2018-04-02:
In the meantime, I tried out an Apple iPad charger with a micro USB cable. This charger provides 2.4 amps, above the 2.0 amps that the typical charger I had historically used for testing on older raspberry pi units.

I can confirm that with this 2.4 A power supply, I have not had a *single crash* in an entire evening of operation, more or less confirming that this was a power issue.

With that said, it would appear that the image provided above is a functioning 3 B+ image. It includes about a year's worth of LibreELEC fixes and improvements as well.

I'm going to leave this issue open until we cut an official release, but I think it is fair to say that the issue of support for the 3 B+ has been addressed for anyone with a 2.5 A power adapter.

ambreswe2, 2018-04-04:
Accidentally upgraded and that breaks it all so, don’t ;) rainbow screen after that. Will put the other image back.

WanaGo, 2018-04-04:
Hey. I tried the pre-release above, and got through the setup and pre-cache no problem, and then we started watching normal TV for a while. After about 30 mins, the TV automatically changed channel to the HDMI as the Pi seemed to have restarted, and we were left with a crash number etc. Just FYI.

dalehamel, 2018-04-04:
@WanaGo can you confirm your power adapter has *at least* 2.5 A?

WanaGo, 2018-04-04:
Yes, its a 2.5A 5V phone charger. Ill try another one I have which I got from the PiHut a while ago, I think that is also at least 2.5A

WanaGo, 2018-04-04:
My mistake, it was 5V 2.1A. I have just put together a PSU, 5V 3A switchmode brick I had, and wired it up to a micro USB. Just powered it up with that, and will leave it running and see how it works. Ill report back.

Diomet, 2018-04-05:
:+1: 
Installed and used for about 4hours with a generic 2.5A 5V PSU. No problems so far. Only the update notification is annoying :)

WanaGo, 2018-04-06:
Was working all day and then it decided to update itself and it died. Gaa. Will reflash it and do more testing, and disable automatic updates.

dalehamel, 2018-04-06:
I don't believe it should be configured to automatically update, this is a
bit bizarre.

WanaGo, 2018-04-06:
I just reflashed the card, did the precache, and went straight into settings, and this is what it is by default:
[image]

epiclulz666, 2018-04-07:
i can confirm what @WanaGo  is saying, using the same img posted above, mine was set to auto update as well on first bootup. it was acturally trying to download the update as i went in to untick it i had to untick and then reboot the pi to stop it from finishing the DL and then applying the update, i can only gather that if that was to happen the whole pi would crash. i can confrim my pi has been stable for about 8 hours of heavy use without any crashes so far, even using the aeon nox theme as well.

[image]

WanaGo, 2018-04-12:
Mine has been running for a week now, used daily, no crashes after changing the power supply to 5V 3A and disabling the automatic updates.

epiclulz666, 2018-04-12:
I'm using a google home mini power supply and cable I think its about 5V 1.8a with no reboots or crashes. strangely if i use another power supply that was working fine with nemslinux with rasplex the pi goes full retard and reboots it self constantly showing the lighting bolt icon up the top

nunofarruca, 2018-04-12:
@WanaGo @epiclulz666  Using any of the available skins? (aeon or black edition)?

epiclulz666, 2018-04-12:
@nunofarruca i myself have been using aeon nox without any noticeable lag or issues

WanaGo, 2018-04-13:
@nunofarruca I have only been using the default.

I have noticed on this version of Rasplex, the Shared Content is back to a menu of its own. Whereas the current main image has the shared content showing up as Movies, TV Shows etc under my local ones. I assume this is just because this image is a step backwards at the moment?

lochstar, 2018-04-16:
Running stable on my 3 B+. Only issue I'm having is upon first boot the WiFi is sometimes not connecting. I have to manually connect. Anything I can do to help debug?

epiclulz666, 2018-04-16:
@lochstar can't say I have seen that, have you tried turning off Ethernet in settings and just having WiFi set ? By Default I think it leaves Ethernet on, reboot with that set and see if it auto connects to WiFi

lochstar, 2018-04-16:
@epiclulz666 Thanks for the suggestion. I'll try disabling ethernet and see if it makes a difference.

Joshndroid, 2018-05-19:
I've been running the test build for a week or so now with no real issues... i did forget to untick the auto update check which lead to a corruption and a re-install but that was my own stupidity not necessarily an issue of the build.. good job thus far

Zonediver, 2018-05-26:
FYI (Regarding power supply)

1) For "all" Raspberries: DONT USE a smartphon/tablet-charger! 
2) A charges IS NOT a power-supply

Reason:
A charger is designed to provide maximum current and DOESN'T HAVE a stabilised output voltage!
The Voltage of a charger can fluctuate between 4,6V and 5,5V >> very bad for the Pi!
The Pi is a computer-system WHICH NEED a STABILISED voltage of 5,05V - 5,2V

My Pi3+ is running stable with a self-built power supply with a stabilised output voltage of 5,05V an a max. current of 1000mA.

Remember: You need a power-supply and not a charger for the pi.

Zonediver, 2018-05-30:
@ Derkades:
to a) wrong - a charger isnt a power supply.
A power-supply delivers current with "stabilized" Voltage, a charger delivers max. current with fluctuating voltage, because the stabilisation is done at the electronic within the smartphone/tablet.
read this article https://raspberrypi.stackexchange.com/questions/51615/raspberry-pi-power-limitations
and see the Specs of the Pi itself.

HydriaOne, 2018-06-02:
Any news  @dalehamel

Sorentoman, 2018-06-02:
Have been running Dale's build for a couple of weeks now and have had only one glitch when I paused a movie and returned some time later to a unresponsive dark screen. Rebooted and was back in action. Unchecked update setting as stated in earlier post. Glad to have my B+ up and running. My thanks to those who helped get us something to use until a formal release is available.

niXta1, 2018-06-03:
Any news on an official version?

dalehamel, 2018-06-03:
Sorry for the delay on the official version folks, I'm in the middle of moving and it's been a hectic couple of months. We have some good builds, but I haven't had enough time to test something thoroughly enough to be comfortable releasing it.

Please sit tight, I'll get it out as soon as I can.

djhxxx, 2018-06-06:
Could we please have a link to something that will boot so we can checkout our system?

dalehamel, 2018-06-06:
There is a build that should be working posted above

yamahor, 2018-06-13:
so, dalehamel, just to make sure. The link above that was said to be unstable was found to be power supply related, correct? Sorry, new to raspberry pi and ode working in general, setting mine up tonight to run plex

Diomet, 2018-06-14:
After about 2 months running without problems, started to have moments with black screen followed sometimes by no sound. I didn't made any changes.

CriticalPoint, 2018-06-29:
@dalehamel how'd the move go? If you need any help testing a newer build, I'd be happy to help?

CriticalPoint, 2018-06-29:
@yamahor basically yes. It's a 1.8.0.0 build, where the latest is 1.8.0.148. Deploy it and ensure that automatic updating is DISABLED (it _should_ be already, but apparently not in some cases), and feedback any issues you encounter.

dalehamel, 2018-06-29:
Hi folks,

Going to try and get a build going here soon and up for release.

Sorry for the delay... sometimes life gets in the way of hobbies.

lochstar, 2018-07-11:
I've been running this build for a while now, I originally ran in to problems with all settings including login data being wiped on boot. I ended up putting in a newer, larger SD card and doing a fresh install to resolve.

The problem has just come up again, after a little investigation, it appears the `/storage/.cache/cores` dir is using a whopping 13.4GB!

The directory is full of files named `core.!usr!lib!plexht!plexhometheater.x`. Any ideas what's going wrong here?

Baschtlwaschtl, 2018-07-13:
i have this too. 
i wrote an autostart script to delete the cores Folder and the .plexht/temp/Crashreports Folder.

Vanos36, 2018-07-15:
Hello. Can I ask how the progress is going on getting a release for the pi 3b+ I am unable to get it going using any of the versions

[17 more replies]