RasPlex/RasPlex #611: Unable to connect after server / plex upgrade

xenofree, 2018-07-07:
Hi,

Server : FreeBSD 11.2
Plex : plexmediaserver-1.13.2.5154
Pi : Model 3B 
Rasplex :Linux RasPlex 4.9.13 Version: 1.8.0.148-573b6d73
Network : ethernet

12:41:45 T:1733293024   ERROR: CCurlFile::Open failed with code 0 for https://plex.reve.space:32400/?X-Plex-Token=SECRETSTUFF
12:41:45 T:1800401888   ERROR: CCurlFile::FillBuffer - Failed: SSL peer certificate or SSH remote key was not OK(51)

Connectivity is good, the android app is working and the web app is working.
But the server doesn't appear in rasplex.

Thanks for your help.

Regards,

WebSpider, 2018-07-07:
The certificate at plex.reve.space is expired, invalid or untrusted.

xenofree, 2018-07-07:
The certificate has not expired and he is valid.

        Issuer: C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
        Validity
            Not Before: Jun 25 06:14:26 2018 GMT
            Not After : Sep 23 06:14:26 2018 GMT
        Subject: CN=plex.reve.space

WebSpider, 2018-07-07:
So, then I guess it's a question if Let's Encrypt is trusted by rasplex :)

WebSpider, 2018-09-19:
@TEfjLONjDON I can play plex cloud content from rasplex and plex mediaplayer.