RasPlex/RasPlex #613: RasPlex not showing up to fling to

Maxakipust, 2018-08-10:
Hi,
When trying to fling content from any device to my RasPlex, I am unable to see it as a device to fling content to. I have my Raspberry Pi 3 model B connected to ethernet, and my other devices connected to wifi on the same network. 

It would also be nice to have an easy way to get the Mac address of the Pi. The only way I could figure out how to get it is to ssh into the pi and get it from there.

Thanks,
Max

timinski, 2018-09-25:
@Maxakipust 
Same issue here with Pi3B+. Because of PMP / Rasplex issues with the B+, I had to go the route of Raspbian with Kodi and then the Plex plugin for Kodi.  Has been a royal pain to set up.  Works fine as a standalone unit playing music from the server but nada as a target for flinging content from Android.

Did you make any progress?

timinski, 2018-09-26:
Thanks very much for the reply.  I actually got it -- meaning Plex on the Pi3B+ -- by way of installing the latest PMP Android app update.
I.e.,

- install Raspbian
- install Kodi
- install Plex plugin for Kodi
- install latest PMP Android app update (that actually appeared on my Android device yesterday)

I now have the joy of playing my music from my Linux Plex server to the Pi (with HiFiBerry Amp2) via fling and control from my mobile phone.  The sound quality is awesome.

spicerackk, 2019-09-10:
Have this issue as well. Sometimes a reboot fixes it, and sometimes refreshing and reconnecting the ethernet fixes it, but then after roughly 2 minutes the RasPi is gone from the clients list. I need it to be an "always on" client for another project I use it with.

Have done multiple fresh installs, nothing changes.

bt6946, 2022-02-06:
I've also experienced this and I'm wondering if it's the case for anyone else (or if this is just a totally new issue) but when I'm using a different home user (i.e. when my partner wants to play music on the RasPlex via their phone) it doesn't show up at all. If I switch their device to my home user, it does show up. 

If anyone knows of a way around that, I'd be grateful...