RasPlex/RasPlex #619: Hi Is this project dead?

bornlibra23, 2018-10-10:
There hasnt been a commit in over two years.

closed 2018-10-10 by dalehamel

4oo4, 2018-10-14:
Could we please get some more info on where current development is at? Closing the issue seems to say that it's still under development, but not giving us any comment doesn't tell us anything.

I haven't run into any major issues running the latest release, but it's good for the community to know what to expect for the future of the project and for responsiveness if a bug is discovered later on.

dalehamel, 2018-10-14:
> Could we please get some more info on where current development is at?

Alright, I'll post this here but then I'm locking this thread. I closed this issue as a duplicate, as it has come up before. Development is not dead, but is in a bit of an intermission. This has happened during the RasPlex project before, which has been going for almost 6 years now. This is a long time to maintain a project, and lots of developers have come and gone over the years.

Plex Inc has created their own raspberry pi plex solution, so they aren't contributing to Plex HT anymore, which is the actual media center application for plex. This has been renamed to "OpenPHT", and the fork is currently maintained by @Kwiboo primarily, as far as I know. He hosts the update server, and currently has the keys to the release pipeline. He is currently running the CI system, and has folded support for RasPlex in with a number of other OpenELEC targets. Current open source plex player development is now mostly on [plex media player](https://github.com/plexinc/plex-media-player), which is the same team from plex that maintained plex-home-theater. OpenPHT is ultimately based on the XBMC media player, but plex-media-player was built from the ground-up by plexinc using its own player framework. I believe this can be compiled for the raspberry pi, but I'm not sure where their latest build is for this.

When plexinc switched to https://www.plex.tv/en-ca/blog/introducing-the-plex-media-player/, the actual funded development for plexht stopped, but the project remains open-sourced as "openpht". 

Both @Kwiboo and I are really busy, we are able to connect once in a while but haven't had time to QA the latest build.

I recently bought a house and have been pre-occupied with renovations and haven't set up my test lab yet, which is the main thing holding up the release.

Before releasing a build, we need to manually run through a number of tests, to make sure that it is compatible with all hardware. 

The main issue with RasPlex right now is [3 B+ support](https://github.com/RasPlex/RasPlex/issues/612), which we have a build for but hasn't gone through QA to be in the official release channel. I don't want people with 3 B+'s to be high and dry, so we cut an early build and released it.

Back when I was working on 3 B+ support, libreELEC (the operating system that we deploy OpenPHT to) was very close to having a release with good 3 B+ support, but the release took too long, and I had to switch to house stuff. This is why there has not been an official release all summer.

I am taking the last week of October off from my job, and I am hoping to get my lab set up then, once that is done I should be able to QA everything. 

For improvements to the plex app itself, it is probably better to take those issues to the OpenPHT community, or see if https://github.com/plexinc/plex-media-player meets your needs. It looks like the latest release, https://downloads.plex.tv/plexmediaplayer-openelec/1.3.11.758-4be89b5c/PlexMediaPlayer-1.3.11.758-4be89b5c.RPi2-arm.img.gz is for Rpi2, which may work on the 3B+ if it uses a recent enough build of openelec, but openelec forked off to libreelec a couple of years ago.

I would love to help get the 3b+ support for OpenPHT released officially, and see about having a dual-boot option with the official plex-media-player image. Personally, I want to make sure that rasplex just always works on the raspberry pi, as I love working with the raspberry pi platform, and love plex. I will continue to keep rasplex alive, and I'll point out that rasplex is still actively downloaded and used, going by the hits to our download and update servers.

The download server for rasplex is monitored and has had 99.95% update over the past year, and still serves requests for many downloads through the installer we provide (which admittedly needs some work, but we still offer instructions for manual downloads and installs which are similar to any other raspberry pi distribution).

dalehamel, 2018-10-14:
> There hasnt been a commit in over two years.

This is not accurate, the latest commit was 6 months ago https://github.com/RasPlex/OpenPHT-Embedded

dalehamel, 2018-10-14:
Linked the wrong issue earlier, correct one is https://github.com/RasPlex/RasPlex/issues/606