RasPlex/RasPlex #621: AirPlay/AirTunes doesn't work

pirate, 2018-11-08:
The [FAQ page](http://www.rasplex.com/docs/faq.html) says that AirPlay Audio streaming should work, but after setting it up the RasPlex isn't showing up in my AirPlay devices list on any of my iOS devices.  I double checked with Bonjour Browser on my Mac and it looks like it's not broadcasting the AirTunes service via Bonjour. 

**RPi model**: `RPi2.arm`
**RasPlex**: `1.8.0.148-673b6d73`
**Network**: `Wifi`

dalehamel, 2018-11-08:
Hi,

Rasplex uses the same libraries as XBMC here. Support for airplay has been spotty over the years, as apple continues to change their protocol. It's possible that updating to a new version will help somewhat, but if current versions of Kodi do not have working airplay, it's going to be extremely difficult for rasplex to.

So, perhaps the readme should be updated - it should work if you have an (unreasonable old) iOS, like iOS 7 and an older apple TV, but that's probably not what you're looking for.

pirate, 2018-11-08:
Yeah I'm on iOS 12.1 and macOS 10.12.  Would it be unreasonable to just add shairport-sync to RasPlex with a single checkbox to enable/disable it in settings?  It seems to be working fine on my other Raspi running Ubuntu.

spacecabbie, 2018-12-18:
Just a quick question i see no setting anywhere to enable airplay? Is it correct to assume its on by default and its just not working atm or am doing something wrong ?

arubis, 2019-01-17:
This functionality was removed: https://github.com/RasPlex/RasPlex/issues/565#issuecomment-289233138