RasPlex/RasPlex #623: Pi 3 B+ packs the cores folder (and other issues)

PastyPilgrim, 2018-11-18:
I know that the pi 3 b+ beta is pre-release and Dale has his hands full, but I just wanted to document my own experience with the pre-release in case it helps with QA.

On a fresh install with a new pi (with proper 5v/2.5a power supply) and a new mSD card, I was experiencing a little bit of quirky behavior including:

- Unable to use in-playback menus (subtitles, audio, etc.). I can access the on-screen display, but if I try to enter any sub menu, that menu would not appear. I can hear the click of going through those menu options, but they are visually absent.

- The seek bar remains on screen permanently. If I ever seek during playback, the seek bar will not go away unless I close the media and reopen it.

- Various crashes and freezes.

- Those crashes generate crash reports and if there is a crash report, rasplex will continuously loop the crash report message on screen until I manually purge the crash reports folder.

- But the biggest issue is that the cores folder is being dumped to continuously. There came a time when I just couldn't use rasplex anymore and I discovered that it was because my mSD card was completely full. Plugging that card into my desktop to diagnose led to tons of issues. Nemo reported that the card had 160GB of content in the cores folder (on a 32GB card =/ ). Trying to delete it then crashed nemo. Eventually I just did a full zeroes overwrite of the card to salvage it.

I can get by with the first couple of issues so going forward I will setup a cronjob to purge the cores folder and the crash reports folder probably hourly to prevent fatal issues.

**RPi model**: 3 b+
**RasPlex**: 1.80 pre-release
**PMS**: 1.14.0.5465 (latest)
**Network**: ethernet

4oo4, 2018-11-29:
@PastyPilgrim I'm also having the same issue with the cores folder filling up, just discovered this today. So far haven't had any of the other issues that you mentioned.

How often do you tell cron to clear that out?

PastyPilgrim, 2018-11-29:
@4oo4 Hourly. Here's the script:

> RasPlex:~ # cat clean.sh 
#!/bin/sh
rm -rf /storage/.cache/cores/*
rm -rf /storage/.plexht/temp/autoupdate/*
rm -rf /storage/.plexht/temp/CrashReports/*

I added autoupdate because even though I have autocheck for updates turned off, it was still checking for, downloading, and trying to install the official release (when only this beta build will work on this pi).

> RasPlex:~ # crontab -e
>0 * * * * /storage/clean.sh