RasPlex/RasPlex #628: Does not boot

andocromn, 2019-02-12:
Not sure what else to say, the 1.8.0 image is not booting on my RPI 2B.  Rainbow screen and nothing else

**RPi model**: 2B
**RasPlex**: 1.8.0
**PMS**: N/A
**Network**: N/A

dalehamel, 2019-02-12:
Have you tried another SD card?

andocromn, 2019-02-13:
Yeah tried 2 SD cards, both work with raspbian.  I did get it working with the image for RPI v1, but the image for RPI V2 doesn't seem to work on the RPI v2

dalehamel, 2019-02-13:
It may draw more power than other distros, I forget what the detail power
management options are.

Did you try a different power source? 2.0A ? Do you have the official
raspberry pi one to test against?

andocromn, 2019-02-13:
I do not have an "official" power supply but I do have many 2.0A supplies.  I can try a different one tomorrow, however I sincerely doubt this is the problem since its not even getting the kernel loaded

chrisherman, 2020-05-09:
tried several times and the same HW works perfectly fine with raspbian and ubuntu. sth. wrong with the image.

GregariousJB, 2021-06-01:
My bad for the major necro, but this popped up in a Google search so it might help someone else.

**The RasPlex installer is completely useless** if you have a Pi 2 or Pi 3. Select either the Pi 1 or Pi 2 option in the installer and you'll still download the Pi 1 image either way, so you go to write it on your SD card thinking it must be the correct one and you get the rainbow screen.

I had to find this fix in [a year-old reddit thread](https://www.reddit.com/r/PleX/comments/cr6p74/best_option_for_plex_client_on_raspberry_pi_3b/ex2ajsl/) to fix it for myself, posted because another person was having the same problem. You can get the Pi 2+ image (which also seems to work-ish for Pi 3) here:  https://github.com/RasPlex/RasPlex/releases/download/PRE-1.8.0b%2B/RasPlex-1.8.0.0-573b6d73-RPi2.arm.img.gz

Furthermore, the installer doesn't mention that the "Pi 2" version is apparently also compatible with Pi 3, which was requested by someone _[4 years ago](https://github.com/RasPlex/rasplex-installer/issues/41)_.