RasPlex/RasPlex #630: RasPlex for PI 3b+

ov10fac, 2019-03-27:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: [raspberry pi model]
**RasPlex**: [version and skin]
**PMS**: [version and operating system]
**Network**: [ethernet/wifi]

ov10fac, 2019-03-27:
This is my first issue so excuse me if I am repeating something or doing something wrong.
I have seen many issues about RasPlex for the Pi 3B+ and some indicate that it has been resolved, some that it is still open.  All issues are closed so...
I have a raspberry Pi 3B+ and RasPlex-1.8.0.148-573b6d73-RPi2.arm is not working for me. 
First, running a Pi 3B+ as I said connected to both wifi and ethernet.  My micro SDcard is a SanDisk Ultra Plus UHS-1 card 32GB brand new.
My Pi is also brand new.
I can run Rasbian and OSMC just fine on the new Pi.  However, when I burn the above img to the SDcard and try a boot all I get is rainbow screen and lightning bolt.  According the the Pi manufacturer that indicates incompatible software.  I only have the one Pi so can't check it against another machine.
If anyone has any ideas how to get it running I would very much appreciate the help.

derJan2001, 2019-03-28:
Sorry, I found no answers.
Why all Pi3B+ issues are closed in a few hours?
Why is there no link on plex.tv to rasplex anymore?

Thanks for your help.
BR Jan

mplinuxgeek, 2019-04-10:
@derJan2001 I find it astonishing that the current model Pi released 12 months ago is still unsupported... and that the current stable release is from 2 years ago, with an unstable Pi 3+ release that is 12 months old. I hate when projects go down this path, I appreciate that the devs put so much effort into it getting a project this far but doesn't mean I have to like it when they go cold like this.

dalehamel, 2019-04-10:
It’s volunteer work my friend @mplinuxgeek

We have lives outside of the project and it gets in the way sometimes, I’m sure you can empathize.

mplinuxgeek, 2019-04-10:
> It’s volunteer work my friend @mplinuxgeek
> 
> We have lives outside of the project and it gets in the way sometimes, I’m sure you can empathize.

Like I said, I appreciate, doesn't mean I have to like it.

dalehamel, 2019-04-10:
There have been several instances where I’ve tried to set aside the time to cut a proper release, but work and my personal life, and other open source projects have gotten in the way.

When I have the time and mental energy a release will be cut and tested. I’m sorry it has taken a year.

mplinuxgeek, 2019-04-10:
Understood... Still don't like it ;)

ov10fac, 2019-04-11:
I can appreciate the conflict between work, personal life and volunteer work.  There are alternates to RasPlex, probably not as elegant, but they work for the time until RasPlex is "Ready for Prime Time".  Thanks for the work you have already done and for the work you will do in the future.

picklewill, 2019-04-15:
Maybe I'm on the wrong thread, but I too appreciate all the work and effort that goes into creating and maintaining the Rasplex project. I have no current issue with the release I'm running [Version 1.15.2.793] with a Raspberrypi B+. Apparently the repository no longer exists when trying to perform and upgrade, and maybe I just haven't searched forums enough to find the new repository, but I'm a copy / paste kinda programmer =P

To speak on OPs issue @ov10fac I hope you are able to get your Pi up and running. This wasn't the video I used to follow the setup, but maybe it might work for you.
[https://youtu.be/3OSnDc0OSto]

Anyway, thanks again to the devs for all the hard work. You may go and live your lives now.

picklewill, 2019-04-15:
@ov10fac This was the youtube video I used to setup my Rasplex. The description section of the video also has majority of the commands needed.  Hopefully this repository will still allow you to get it up and running

https://youtu.be/v0r7daykHAw

dalehamel, 2019-04-17:
As I mention in https://github.com/RasPlex/OpenPHT/pull/315 I am working on a new build of rasplex that will fix a timeline saving issue, as well as likely bring 3B+ support.

I will try and remember to post some build links here once I have something up and running, for anyone here to try out.

dalehamel, 2019-04-17:
And here's the branch i'm working off of for LibreELEC https://github.com/RasPlex/OpenPHT-Embedded/pull/2 as I mention, the 3b+ experience should be identical to that of LibreELEC 9.0.1, as it will use the same kernel and base operating system, built with the same toolchain.

dalehamel, 2019-04-18:
The image i've built so far boots in my qemu test, but i'm already aware of some limitations. It'll be a few days at least before i have something worth testing here.

WanaGo, 2019-07-03:
Hi @dalehamel 
Bringing this up again. Sorry.
3B+, is there going to be an official release anytime soon? 
Pi4, will this be included or is it so different it needs time to develop that...?

The beta release you gave about a year ago now, has never worked reliably for me, I have tried about 2 dozen times and no matter what I do it results in corruption or crashing. I am back to using the Pi2.

I know you have your own life and this is non-paid, etc etc. I'm sure we all appreciate that. But this has been going on for a ridiculous amount of time.
I have been using Rasplex since the Pi1 and have 7 installations of it running and used every day. I do appreciate your efforts, don't get me wrong, but I feel like we are being strung along waiting for these updates and they never seem to be happening.

PLEASE can you release a proper release for the 3B+ and now the Pi4 too... ??

VersatileNinja, 2019-07-13:
Love to hear update from dev what current status is

cburkins, 2019-07-24:
Just wanted to say thanks to @dalehamel .  His recent (unofficial) build for Raspberry Pi 3 is running just fine for me.   Minimal testing (watched a Star Wars movie :), but seems to be working great.   Looks like you've been successfully supporting Plex on the Pi for a number of years. Thanks very much for that.

WanaGo, 2019-07-24:
> Just wanted to say thanks to @dalehamel . His recent (unofficial) build for Raspberry Pi 3 is running just fine for me. Minimal testing (watched a Star Wars movie :), but seems to be working great. [...]

You are referring to the beta release, for Pi 3B+ I assume. Not Pi 3. Pi3 runs fine on the main image, a beta is not required for that. The beta was to make the 3B+ work...

cburkins, 2019-07-24:
Precisely correct, thank-you for the correction !  Yes, I'm running the beta release on a Pi 3B+.   (stable release gave me the same permanent rainbow screen that others experienced).

sgabe, 2019-08-06:
I can confirm that the beta release seems to be stable on a Pi 3B+ though I have encountered some sound distortion issues but that is another topic. The background music in some scenes reminds me to an old walkman with a dying battery.

aniel300, 2019-08-16:
any update or test build for pi 3 and/or 4 ?

AKDub, 2019-08-16:
> any update or test build for pi 3 and/or 4 ? and where can i get it?

All versions of 3 are working with at least the beta image in the releases page.

I think like many of us you will have to be patient for any 4 version that may come down the line

aniel300, 2019-08-16:
do u have a link for the beta image and do you know if the pi 3 is capable of 265 10bit direct playback ?

AKDub, 2019-08-16:
https://github.com/RasPlex/RasPlex/releases/download/PRE-1.8.0b%2B/RasPlex-1.8.0.0-573b6d73-RPi2.arm.img.gz

Maybe at lower resolutions like 720p, but the main reason I am hoping for a pi4 version is for 4K H.265 playback

aniel300, 2019-08-16:
that is the same image i used..it has a limitation of 1080 265 8bit i guess i will give a try to pi 2 or something else. i wonder if pi 3 will playback 10bit by using libreelec ?

BinTechLLC, 2019-08-16:
UNSUBSCRIBE

-- 
~~~~~~~~~~~~~~~~~~~~
Koltin Binford

-Owner
-Binford Integrated Technology

Office:  (720) 515-8181
Mobile: (720) 492-3881

*CONFIDENTIALITY NOTICE:*
The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then delete this message and any attachments. If you are
not the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments is
strictly prohibited.

~~~~~~~~~~~~~~~~~~~~