RasPlex/RasPlex #631: RasPlex 1.8.0 not saving play status in PMS 1.15.3.876

trumpy81, 2019-04-05:
After watching a TV Show or Movie, if playback is stopped at any point, instead of saving the play state, Plex will immediately revert to an unplayed state.

I have verified that the play state is saved correctly when using Plex/Web, but not when using RasPlex on the RPi2.

The problem started with PMS version 1.15.2.793, so I guess something has changed in PMS?

**RPi model**: RPi 2 or RPi 3B (not plus)
**RasPlex**: 1.8.0 default skin
**PMS**: PlexMediaServer-1.15.3.876-ad6e39743 on Synology or Windows 10
**Network**: WiFi

trumpy81, 2019-04-09:
For some strange unknown reason, this has miraculously started working.

Beats me why it was not working. I can only imagine there was some kind of network issue preventing things from communicating correctly.

Anyhow, alls well that ends well. :)

closed 2019-04-09 by trumpy81

Guy0me, 2019-04-12:
Hello i have the same issue on RPi3 since the version 1.15.3 of Plex Media Server, TV Show or Movie, or anything else.. if playback is stopped at any point, instead of saving the play state its start from the beginning.
I do the same action on a plex apps on Hisense TV there is no issue.

ArnieLister, 2019-04-12:
I'm experiencing the same problem since server updated to 1.15.3.876.  Confirmed on 3 pi's (2 RPi3 and 1 RPi2) each with separate RasPlex installations.  PMP embedded for Pi (2.31 nightly build) does not have this issue, strongly indicates an incompatibility has been introduced specific to RasPlex.

dalehamel, 2019-04-12:
I have noticed this issue specifically with transcoded content.

With content that can be played back natively on the raspberry pi, this issue doesn't seem to occur.

> PMP embedded for Pi (2.31 nightly build) does not have this issue, strongly indicates an incompatibility has been introduced specific to RasPlex.

PMP is a newer codebase, and may have newer handling for such errors with transcoding.

Has anyone noticed this issue with *native* playback? Eg, you don't see a transcoding window? Do you still experience the issue after restarting the plex server process, and rebooting the raspberry pi?

ArnieLister, 2019-04-13:
This topic has been more widely discussed for OpenPHT here with responses from Plex employees:

https://forums.plex.tv/t/lasts-2-versions-broke-save-resume-opht/396505

Plex are laying the fault with long standing bugs with OpenPHT (and therefore RasPlex) reporting play progress incorrectly to the server.  Latest PMS updates have become less tolerant of these bugs in an effort to improve resume performance in other apps.  AFAIK OpenPHT isn't being developed anymore, so this could be a serious nail in its coffin.

dalehamel, 2019-04-13:
@arnielister I’ll take a look when I have the chance. I haven’t seen the issue with non-transcoded content, even on the latest server version.

If the bugs are well defined I’m sure it can be patched, but I can’t fix what I can’t reproduce

ArnieLister, 2019-04-14:
@dalehamel I can confirm that I get the problem when playing non-transcoded original quality direct-play content. RasPlex is still far nicer and way more responsive UI experience than PMP embedded, but this is a show stopper for me.

dalehamel, 2019-04-15:
I can confirm I can reproduce this.

I’ll look into what code needs to be changed, but if you’re aware of any forum posts that describe the API issue it would make it easier to fix

ArnieLister, 2019-04-15:
This comment by plex employee gbooker02 refers to an OPHT bug in the resume process:

https://forums.plex.tv/t/lasts-2-versions-broke-save-resume-opht/396505/4

paul-bsc, 2019-04-16:
Seeing the same behavior my side. Really wish I could help to try and revive this. RasPlex is by and far the **best** Plex client. 

The following two commits could at least maybe provide some clues?

This one seems to touch most of the resume process: https://github.com/RasPlex/OpenPHT/commit/6791924b4686a24b7cfa4d4518b8143bc61da80f

This one specifically looks at transcoding resume:
https://github.com/RasPlex/OpenPHT/commit/916f34a4a2b14cdd7ad1c5fff72b83af33244272

This file seems to be the best lead given the comment made on the Plex forum: https://github.com/RasPlex/OpenPHT/blob/60b75e1ab50495e28b76c8f2959fba3f409bcb8b/xbmc/utils/SaveFileStateJob.h

But for the life of me I can't find any solid reference to the API changes in PMS - probably just not looking in the right places.

dalehamel, 2019-04-16:
Thank you, these links are helpful. 

I will also look at plex media player to compare what it’s source code does differently for saving. Although it uses different frameworks (I believe it’s Qt instead of Boost), it should clearly show the proper client behaviour. 

I recently updated my own server and started experiencing this issue as well. Since it is annoying me directly I have my own motivations to fix it ;)

I’m coming up on having a month off for the first time in ages, no promises but this is on my list of things to do (as well as getting a 3B+ build out).

paul-bsc, 2019-04-16:
Excellent! I'm also happy to try and get the dev build process working on my side to help where I can, but judging from comments on other issues raised it looks like it may be tricky at this stage. I will make time in the next few weeks to see if I can get that working locally.

paul-bsc, 2019-04-16:
[This](https://github.com/plexinc/plex-media-player/commit/304f4da215f2c050e9618b94c18c0ef55a6cc962) looks like the correct commit to reference in PMP.

dalehamel, 2019-04-16:
The corresponding code in openpht appears to live in https://github.com/RasPlex/OpenPHT/blob/openpht-1.9/plex/Client/PlexTimelineManager.cpp

dalehamel, 2019-04-16:
There appear to be a number of lovely log statements peppered throughout the relevant code here.

If someone has a moment, it would be helpful to see the logs of it failing to update the play progress. I imagine it will give some helpful clues.

@paul-bsc if what you indicated, then they just added an api to distinguish between "stopped" and "finished/canceled". If that's the case, the change might be trivial to support.

dalehamel, 2019-04-16:
https://github.com/RasPlex/OpenPHT/blob/openpht-1.9/plex/Client/PlexTimelineManager.cpp#L251-L255

https://github.com/RasPlex/OpenPHT/blob/60b75e1ab50495e28b76c8f2959fba3f409bcb8b/plex/PlexTypes.h#L69-L74

That enum probably needs to be updated to match  https://github.com/plexinc/plex-media-player/blob/304f4da215f2c050e9618b94c18c0ef55a6cc962/src/player/PlayerComponent.h#L130-L137 and https://github.com/RasPlex/OpenPHT/blob/60b75e1ab50495e28b76c8f2959fba3f409bcb8b/plex/PlexUtils.cpp#L783-L800 to get the string representation.

This is what is used to build HTTP request that is sent to the plex server. Those changes to reflect the changes in play states made and expected upstream are certainly needed. From my read of that PR, they ignore the "stopped" state, so setting this to "cancelled" would likely get the desired behavior to work again quite quickly.

beyond that, additional handling to ensure that finish and canceled are handled appropriately is probably also needed.

dalehamel, 2019-04-17:
Started working on https://github.com/RasPlex/OpenPHT/pull/315 to address this issue

snix47, 2019-05-19:
I have the same problem, rasplex does not save play status since I upgraded plex server to latest version.

dalehamel, 2019-05-19:
Plex folks have said they’ll be fixing it, and it’ll be when the next release they cut is out.

I tried the latest server version as of May 15 and it’s not yet landed in that.

The plex forum post describes the API issue nicely, so I can update my patch accordingly.

Have a build that might fix it but not yet tested.

wurlyfan, 2019-06-08:
I have same problem, and have switched to PMP embedded for Pi as a temporary solution. Keen to get back to RasPlex, though, as soon as this is issue is resolved.

lochstar, 2019-06-08:
Beta PMS version 1.15.8.1198 landed yesterday.

>Certain clients (third party Samsung and OpenPHT) would fail to resume video.

wurlyfan, 2019-06-10:
I can't see this latest version - does it address the play status issue for RasPlex?

dalehamel, 2019-06-10:
Yes I can confirm the server release mentioned (in the beta channel, available to plex pass holders for now, generally available at a later date I believe) fixes the issue for me.

I still intend to finish the PR i mentioned earlier that should update the API, allowing rasplex to work with any server version without needing this special quirk handling.

In the meantime, the updated server is confirmed to work around this issue for rasplex.

Guy0me, 2019-06-11:
So now what do we have to do to get it working ? 
an update is it previewed on Rasplex automatically on server side or should it be an update on client side ? i mean on the raspberry directly ?
Thanks in advance.

lochstar, 2019-06-12:
@Guy0me Automatically on the server side. Hopefully @dalehamel will have a Rasplex update soon that will prevent it coming up again in the future.

dalehamel, 2019-07-15:
I've been able to confirm that the fix that Plex released to their server resolves this issue for now.

I intend to get to this in order to be compliant with the new API, but it's a higher priority to get the latest LibreELEC pulled in, as that will be what brings support for #630 and #635, and will allow for continued support of RasPlex on future versions of the Raspberry Pi.

Hopefully the patched server is generally available already or will be soon