RasPlex/RasPlex #634: Vagrant Setup

wpmccormick, 2019-05-19:
Running VirtualBox 5.2.18; vagrant 2.0.2 on Ubuntu 18.04, following the vagrant setup instructions [here](https://github.com/RasPlex/RasPlex/wiki/Vagrant-Setup), everything works fine until I run buildman. 

This sequence is repeated several times:
`
--2019-05-19 18:15:31--  http://samba.org/ftp/ccache/ccache-3.1.8.tar.bz2
Resolving samba.org (samba.org)... 144.76.82.156, 2a01:4f8:192:486::443:2
Connecting to samba.org (samba.org)|144.76.82.156|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://samba.org/ftp/ccache/ccache-3.1.8.tar.bz2 [following]
--2019-05-19 18:15:31--  https://samba.org/ftp/ccache/ccache-3.1.8.tar.bz2
Connecting to samba.org (samba.org)|144.76.82.156|:443... connected.
OpenSSL: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version
Unable to establish SSL connection.
--2019-05-19 18:15:31--  http://sources.rasplex.com/mirror/ccache/ccache-3.1.8.tar.bz2
Resolving sources.rasplex.com (sources.rasplex.com)... 162.243.195.16
Connecting to sources.rasplex.com (sources.rasplex.com)|162.243.195.16|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-05-19 18:15:32 ERROR 404: Not Found.
`
Followed by:
`
Cant't get ccache sources : http://samba.org/ftp/ccache/ccache-3.1.8.tar.bz2
 Try later !!
make: *** [release] Error 1
Request processed in 26.98s
`

My goal is to get a build environment setup and hopefully begin making some contribution.

Cheers!

wpmccormick, 2019-05-19:
I've had success building 1.8 from a generic Ubuntu 14.04 vagrant with cpus=4 and memory=8192.

wpmccormick, 2019-05-21:
After switching to branch openpht-1.9 and branch timeline-fix:

```
make[4]: Entering directory '/home/vagrant/RasPlex/OpenELEC.tv/build.RasPlex-RPi2.arm-8.2-devel/openssl-1.0.2o/.x86_64-linux-gnu/engines/ccgost'
../../libcrypto.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
make[4]: *** [../../Makefile.shared:167: link_o.gnu] Error 1
```

dalehamel, 2019-07-15:
Hey,

The build process is much easier with docker https://github.com/RasPlex/Dockerfiles

This was the branch that @Kwiboo left off, and what I have been using for my latest builds to try and Rebase the upstream LibreELEC to add support for 3b+ and 4