RasPlex/RasPlex #635: Rasplex build for Raspberry Pi 4?

Cossack1977, 2019-06-24:
With the Release of the Pi4 is this enough to get the dev's back on board? if so i'm ordering 6 boards now.

WanaGo, 2019-06-24:
Came on to ask exactly the same thing. I still cannot get the Pi3 running stable on the beta release :(

lessthantom, 2019-07-11:
> Came on to ask exactly the same thing. I still cannot get the Pi3B+ running stable on the beta release :(

works fine for me on my Pi 3B+

VersatileNinja, 2019-07-13:
Please advise working 3b+ version

lessthantom, 2019-07-14:
I dont know what i can advise really it just kinda works i didnt do anything special used the installer for the pi 2 and it works......

Not sure if its missing any features it should have but it plays everything i ask it to play

WanaGo, 2019-07-14:
@lessthantom Are you use your on a 3B+ and not a 3B?
I can't even get the latest standard release to boot on the 2x 3B+ I have. There was a beta released (you will need to check some of the old issue posts @VersatileNinja), which did work for a few days, but it tries to update itself which corrupts itself so you have to disable that, and then it does lots of logging or something and fills the card up. But it just goes totally unstable for me after about 3 days, to the point where you have to reflash the card. This is the beta version I am talking about.

dalehamel, 2019-07-15:
Hi folks,

Sorry for the delayed reply.

Yup, I plan to add support to the raspberry pi v4.

I couple of months ago i rebased the underlying operating system (libreelec.tv) and still had a few conflicts to resolve:
- The python that OpenPHT (an old fork of XBMC) wasn't linking to the system python, so the startup dialog wasn't working. 
- OpenPHT was starting, but was crashing while trying to initialize something (probably python)

These were on 3B+, and a rebase onto the branch of LibreELEC that supports raspberry Pi v4 would probably not be too complicated, as most of the updates are probably in the linux kernel.

I believe the kernel, which was a pre-release kernel for the build I cut of 3b+ a little over a year ago, was probably the source of the instability.

I will admit that I haven't been able to put much of my "OpenSource" time into RasPlex lately, I've been busy working on iovisor/bpftrace and some other work that is more closely related to by day job. However, I've always wanted to apply something like bpftrace to RasPlex, to help optimize it and make it faster and more stable.

Please understand that maintaining this project is a passion project, and you might have to be patient to get a new release. The build infrastructure we used to have for rasplex is no longer around, as other developers have moved on to other things, so releases must be cut by hand. I know it sucks to wait, and I feel bad that the latest hardware isn't supported (and hasn't been for a little over a year). 

For myself, I still use RasPlex every day and I've been running mostly on a very stable Raspberry Pi 3 build that was the most recent release. I think I can get something like this out again, but I'll need to get the build up and test it myself for a few weeks to ensure it is stable. If I could have a week of uninterrupted time to revive the RasPlex build, I now have the space for my testing lab, to run it through each raspberry pi board.

When I last rebased LibreELEC, i also was able run the rest via qemu, which will help with future automated build testing here. 

Thanks everyone for staying fans of RasPlex, I hope you don't mind sticking it out a little bit longer. I've been shifting around my work schedule and trying to make more room for fun projects like this.

I'm going to lock this thread until I have something new to report.