RasPlex/RasPlex #636: Rasplex Pi4

apache81, 2019-10-11:
Can anyone let me know do the latest versions of rasplex work on the pi4?

WanaGo, 2019-10-21:
None, still waiting for Pi3B+ stable support, let alone Pi4. :( 
There is a PI3B+ alpha release, some have had luck with it, some havent. Mine I can run for between a week and a month before it self implodes and has endless crashes and corruption. Have 2 Pi4's sitting here waiting to use them for Rasplex, but no software to run on them as of yet.

mrjims1, 2019-10-22:
Not Rasplex but I've gone down this route on my Pi4
https://tynick.com/blog/09-22-2019/plex-media-server-on-raspberry-pi-4/
Runs fine, I can DP UHD and HD streams simultaneously. Also have Sonnar, Raddar and Sabnzb running.

hapklaar, 2019-10-22:
> Not Rasplex but I've gone down this route on my Pi4
> https://tynick.com/blog/09-22-2019/plex-media-server-on-raspberry-pi-4/
> Runs fine, I can DP UHD and HD streams simultaneously. Also have Sonnar, Raddar and Sabnzb running.

That's fine for a server install, but people waitinig for Rasplex just want a simple client. Really hope a RPi4 release is imminent! ;)