RasPlex/RasPlex #638: Great Project, RPi 4 Incoming?

InoSiX, 2020-01-10:
Wanted to start by saying there are few open source projects that I have used that hit it out of the park so consistently as Rasplex has. I LOVE it. I have used it in the past for more than a few folks to help get them connected. It works great and looks very professional.

That being said, as I type this, I just went out to buy a Raspberry Pi 4 to finish a media room project at home. I'm sitting right next to it. I actually had the expectation I would be able to take a spare SD card that had the latest build of Rasplex on it and place it in my RPi 4. I have since discovered it doesn't boot.

I have made note that in mid July you hinted at a potential release that would allow for use on the RPi 4.

Is there any chance we can get this working on the powerhouse that is Rpi 4?

lessthantom, 2020-01-15:
in the July thread Dale locked the thread until he has something new to report. It hasnt been unlocked so that probably answers your question. 

I'm waiting patiently till its here to buy a pi4 as well. But he's a busy guy so we will need to just wait/hope

lessthantom, 2020-01-15:
I guess with anything new technology wise (as in the whole project board thing) that is in real terms in its infancy. The development jumps between 2/3/4 have been pretty big and involved architecture changes which has made simple upgrades trickier

Its a shame and i feel your pain as 100% of me buying a pi4 is for rasplex im just lucky i havent grabbed one yet

WanaGo, 2020-02-24:
@dalehamel Please, any news on Pi4 support (or Pi3B+ for that matter)... ?

rsilvestre, 2020-03-24:
Hey @dalehamel,

I'm still hesitating if I should by the new raspberry pi4B or not, because I'm only using it to watch media movies from my Synology with a plex server on it. The fact is that without rasplex, that don't make any sense. My problem is that my raspberry pi3 starts to become old and I have an issue with it.

Maybe if you need help, some of use could give you assistance ?

Regards,

mistersoft101, 2020-03-24:
Is there a a 100 ppl willing to spends 100euro ?
Or maybe a 1000 ppl willing to spends 100euro on this ?
I would,  At least to have a RPi 4 with 4k output capability, and native
h265 decode & a Stable Rasplex/OpenPHT implementation.

start a crowdfunding page Dale and "they will come"...    but in what
numbers.

lessthantom, 2020-03-24:
Before i say this i’d like to say i love rasplex but 

Why would you pay 100 euros when you can buy an alternative device that does it for already for similar if not less money 

I ended up going to an nvidia shield more than 100 euros i know but it works a treat, i have moved my pi to an alternative none 4k tv which it is fine for 

Im all for crowd funding and would throw a few quid in but i dont think anyone would throw that money in (maybe i am wrong who knows)

hapklaar, 2020-03-24:
@lessthantom I think most people have moved on to other Plex players, like the one on the Shield or on their smart TV. I do however still miss the simple but effective interface Rasplex gave me for years, but also starting to get used to the more 'modern' Plex clients.

lessthantom, 2020-03-24:
@hapklaar 

Agree 100% i moved on to the shield and you certainly get what you pay for, it never misses a beat. Rasplex was/is one of the best integrations

mistersoft101, 2020-03-25:
I haven't tried the Shield ( and I'm hardware agnostic). and 100euro/dollar
(e.g. for a lifetime, multi device license especially) would be fine for me.

Does Shield run an official Plex App (because I've hated both the look &
feel, and lack of customisability with those)
Or an PHT/OpenPHT  offshoot like the RasPlex ?    Ideally I'd just like to
keep that that interface or very similar. but with 4K and native h265
decode.
and willing to pay if the opportunity comes up

That for the replies.  Hopefully Dale or similar guys with the skills &
code access bring Rasplex/Equivalent PHT experience to the RPi 4 in time
anyway -  Thanks Dale for all the previous work to date on Rasplex either
way!!!

lessthantom, 2020-03-25:
@mistersoft101 

The shield using the official plex client but it tuns android so you could i guess use kodi and plexforkodi addon which has a more openpht feel to it. 

You could do the same with all android boxes though so it doesn't need to just be a shield, it does play 4k and h265 out of the box though which is pleasant 

I agree i prefer the old pht interface to the newer plex setup but im used to it now.

mistersoft101, 2020-03-25:
Thank you LessThanTom !

Yes I've tried Kodi+Plexforkodi addon before - but it's not the same as the
OpenPHT experience.
I understand beggars can't be choosers tho :)
Might give it another go though.
In the meantime using only h264/and sticking with 1080p on a RPi2 running
rasplex is hardly the end-of-the-world ;)
On a LG oled still looks fairly cracking

lessthantom, 2020-03-25:
@mistersoft101 yeh i don't think anything really does the openpht interface anymore mainly because plex home theatre hasn't been a thing really for a while now which is a shame 

I expect if we did see a new rasplex it would be need to be based on the new interface, although maybe not or maybe it could be skinned, openpht is based on kodi which is skinnable so i guess maybe you go run plexkodiconnect and skin it if there is a skin that suits? 

Who knows but you are right we cant really be picky haha we got a good few years out of rasplex and its still functioning well

19eighties, 2020-04-19:
I still use this platform. I haven't seen anything that beats the interface. Running from Android is clunky. I'm hoping it can be updated for the Pi4.

InoSiX, 2020-04-23:
> I still use this platform. I haven't seen anything that beats the interface. Running from Android is clunky. I'm hoping it can be updated for the Pi4.

That's my point. It blows everything else out of the water. It looks cleaner, it functions better and it has more options. It is disappointing to find the project in this state. Props to the developer for taking it this far.

rsilvestre, 2020-04-25:
My story with Rpi 4

1. I decided to use Kodi (for the first time). So I installed Kodi and installed the add-on plex player and i observed that the quality of the stream (on the same television than de Rpi3). The quality of the 4k h264 and h265 in general was really bad.

2. After days trying to build plex client for raspberry pi 4, I face crazy difficulty to install QtWebengine. I try a cross build on a debian container running on Docker, because QEmu only support raspberry 1 and 2 and not 3 and 4. By the way, after a couple of time, I decided to build Qt, plex player and dependencies on de raspberry 4 directly with raspbian. After 2 weeks, i finally succeed to build the player.

OMG, the quality was so damned bad ! Drop to the garbage

3. I just bought the raspberry 4 because i expected a better quality in general but it’s not the case. I thought now that I should just stay with my Rpi3 that works good enough for the moment.

Conclusion, don’t buy a Rpi 4 if the only purpose is to what media on plex server with rasplex. The Rpi 3 is enough.

mistersoft101, 2020-04-25:
>    1. I just bought the raspberry 4 because i expected a better quality

mmm is a little sad true.  I'm quite surprised still that their isn't a
little contigent of RasPlex developers that like the 1.8 version, are
generally very happy with it - but would like to add hardware H.265 decode,
and optional 4K output. Hm.
No rush though. Maybe next year or two......

If it happens, great - I'll buy a few Pi4's ; otherwise I'm happy sticking
with my Pi2's (have ethernet).
would willingly pay for such a 1.9/2.0 build though.

uudruid74, 2022-04-10:
Still no RPi 4 release?  There was a bug #619 which seems to say the project isnt dead, but it appears its deader than dead.

I have a pipe dream where my Tidal subscription I get from PlexPass will actually play to my MQA DAC.  I currently use picoreplayer for music and only use Plex on my ChromecastTV which doesn't output bit perfect to my MQA DAC.  I would really like to get the Plex interface married to my DAC and this project was my best hope, but its dead.  So ironic that my Tidal is from PlexPass and they dont actually work together.

Spark-NV, 2022-04-12:
My fork of OpenPHT-Embedded *should* support RPI4 plus a bunch of newer amlogic devices since i rebased it from coreelec 9.2... so it should support all the devices it supports. though i cant test and havnt even attempted to compile any rpi project yet as i cant test them anyways. but i ca confirm that support has been added for newer amlogic devices.

dalehamel, 2022-04-12:
@Spark-NV it is great that you have a rebase of coreelec! 

In theory, only a few packages need updating (the kernel, the firmware usually) in order to support new RPi boards. Often, rebases also introduce instability, so in the past I had favored the approach of cherry-picking the specific packages needed to support a new board, since I haven't had the time to QA new builds in quite a while, which has been the main thing blocking new releases (can't release something if it doesn't work!). 

The PlexHT code itself is long-forked off from XBMC, and that would be a much more painful job to pull forward to the latest Kodi sources, as they are now several years diverged, and Plex Inc isn't supporting that codebase anymore to the best of my knowledge. So, any libraries that it depends on can introduce issues if the base OS update forces them to be updated. In particular, I remember ffmpeg being very finicky, and perhaps even vendored. I have dreams of updating it to the latest Kodi sources someday, but can't make any promises.

If anyone is able to try his build for RPi and it works, please keep this thread updated! I also don't have an RPi 4 to test on either.

Lastly, my apologies for not having the time to keep things up to date on my own, the only contributions I've been able to make lately was the repacking of images when that CA cert went out of date unfortunately.

Spark-NV, 2022-04-12:
> @Spark-NV it is great that you have a rebase of coreelec!
>
> In theory, only a few packages need updating (the kernel, the firmware usually) in order to support new RPi boards. Often, rebases also [...]

Oh yea, I am not at all saying this rebase is stable, I haven't thoroughly tested it. My testing so far is only limited to a S905W box and a S922X box, both of which I've had running for a total of like 20 minutes just to test things are actually working like video playback, wifi, ethernet etc. I haven't done ANY long term testing.

Spark-NV, 2022-04-14:
I have compiled the images that I am capable of compiling and posted them. I had 1 tester using a RPI3b+ say it works for him so I can only assume the rpi devices minus rpi4 work and I have tested 1 Amlogic-ng device(S922X) and 1 Amlogic device(S905w) and it appears to be fine.

As I posted above in my other post that I edited, all the devices that default to the Linux platform config fail to compile with various errors such as missing headers, conflicting definitions etc that I myself am unable to work threw with my very very limited knowledge.

Hopefully someone more knowledgeable than me can get the pi4 and other devices to finish the compile.