RasPlex/RasPlex #641: 1.8.0 install error - rainbow screen

randommonth, 2020-05-09:
Hi Guys,
I've tried to install 1.8.0 on a Raspberry Pi 3B+ and it refuses to perform the initial boot, remaining stuck on the rainbow screen. 
I flashed the card using the Rasplex installer, I'm using a Samsung EVO Plus 32 GB card and the official Raspberry Pi power supply. 
I've also tried about 3 other cards and another 5V 2.5 A power supply but keep seeing the same issue.

Any hints?

**RPi model**: Raspberry Pi 3b+
**RasPlex**: 1.8.0
**PMS**: 
**Network**:

JimPitbar, 2020-05-17:
Hello, I went into the same problem using a Pi 3B+. However I also have a Raspberry Pi 3B which works just fine and my understanding is that 3B+ and above are not (yet) supported.

EgonHanselaer, 2020-06-05:
Hi, it has been a while since this project was updated, For the Raspberry Pi 3B+ there is an (unstable) build that works. 
The author posted it on another issue, you can find the build that does start for the 3B+ here:
[https://github.com/RasPlex/RasPlex/releases/download/PRE-1.8.0b%2B/RasPlex-1.8.0.0-573b6d73-RPi2.arm.img.gz](url)

Spiderstu, 2020-09-24:
I have the same problem with a raspberry pi 2 B. Is there a similar resolution out there somewhere?

pietchaki, 2020-10-19:
> Hi, it has been a while since this project was updated, For the Raspberry Pi 3B+ there is an (unstable) build that works.
> The author posted it on another issue, you can find the build that does [...]

It was released more than 2 years ago, are there any newer release or is there any updates to this project?
It does run smoothly for me in the Pi3B+.

AndrewDiess, 2021-04-03:
For people that are having this issue, the downloader/formatting program downloads the wrong image. It downloads the Raspberry pi 1 image rather than the RPi2 and you need to get the release from the github page instead. I was having this issue myself and noticed the discrepancy in the name of the file. Hope this helps some people.