RasPlex/RasPlex #644: Rasplex - Problem with Photos sorted by cluster

ggmorain, 2021-04-11:
Hi Github Community,

Hereunder, I try to describe as well as possible what I consider as a Rasplex bug.

[Tell us about your issue, steps to reproduce, expected and actual behavior]
Steps to reproduce:
•	First, you must have at least a topic named “Photos” in your media list on the left side of the screen.
•	Then, with your remote control, click on “Photos”. Then a new screen appears with photos ordered by directories or albums depending on your previous choice.
•	On the left side of the screen, you have orange symbols like triangle, crossed arrows…
•	With your remote control, shifting to the left, put the focus on any of these symbols, and then shift again to the left.
•	A new column of textual options appears on the left side of the symbols.
•	Choose “by cluster”. The right side of the screen turns into a list of dates.
•	Then shift the focus to the right. The textual options and the symbols disappear.
•	Now you are into the problem.
•	In this configuration, it is impossible to open any photo from the dates list and it is impossible to leave this mode “by cluster” because nothing appears on the left side of the screen (no symbols and, of course no textual options)

I can go back to other media like films, music.... But when I'm back to Photos, I always have the same screen.

I join a picture of the situation from where I can't escape because there are no more symbols and options on the left end side.
[image]

I also join a picture of the situation in Film mode which looks like it was before in Photos mode
[image]

[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

Other information:
Raspberry Pi 2 Model B V1.1
Type of SD card: SDHC 16 Go class 10 
Power supply type for Rasplex: Aukru Model BS-24W0503000W – Output 5V 3A
Connection: Ethernet
Version of Rasplex: 1.8.0.148-573b6d73, Automatic update

Plex Server: Plex Media Server (v 1.21.2.3943-a91458577) on local NAS Synology (OS: DSM 6.2.4-25556).

Best regards
ggmorain

ggmorain, 2021-04-21:
Nobody can help ?

Maybe I did'nt post in the right forum ?

ggmorain

InoSiX, 2021-04-23:
It's not that you didn't post in the right spot, it's more that you posted in a dead project. Look at the last upload date. It hasn't been, nor is it likely, to ever be updated again. The developer hasn't touched the project in quite some time.

dalehamel, 2021-04-23:
The project hasn’t been updated in some time, life tends to get in the way. I hope to be able to update and release a new version in the future which could address your issues