RasPlex/RasPlex #645: No more access to plex server

sdsOdin, 2021-09-30:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
all RASPLEX clients (in total 7 RPi) in two distinct setups stopped being able to access their respective plex server September 30th roughly 6pm UTC. Plex forum tells me it is related to this 
>> https://forums.plex.tv/t/important-information-about-plex-for-smart-tvs-after-september-30-2021/746506 
although it might be also related to this issue 
>> https://forums.plex.tv/t/pms-suddenly-shows-issues-with-remote-access-9-30-2021-10am-est/747103

I was not able to find a client side setting that would allow the use of unsecured connections to the server. 

Android apps (FireTV, AndroidTV, Android Phone) and browser players work fine.

[Always attempt to reproduce your issue in the default skin] 
did

[Enable debug logging in preferences and insert a link to your log-file]
not quite sure how this would work just yet, but I will try to find out. 

**RPi model**: [raspberry pi model] 
RPi 2 and 3b

**RasPlex**: [version and skin]
1.8.0 default skin

**PMS**: [version and operating system]
Version 1.24.1.4931 (Synology Arm64)

**Network**: [ethernet/wifi]
ethernet

dalehamel, 2021-09-30:
Thank you for reporting the issue @sdsOdin 

I believe you are correct about the issue, it seems to be a problem with SSL connections. I notice this as well on my local media server:

```
...
Sep 30, 2021 20:34:40.321 [0x7f05e5628b38] DEBUG - CERT: incomplete TLS handshake: sslv3 alert certificate expired
Sep 30, 2021 20:34:40.354 [0x7f05e5605b38] DEBUG - CERT: incomplete TLS handshake: sslv3 alert certificate expired
...
```

I suspect it may be related to one of plex's certificates signed by the lets encrypt root cert expiring. Hopefully they can identify the issue and switch to something with a different certificate chain of trust.

BobWhite2017, 2021-09-30:
Everyone will have this issue until certs are updated.

https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

sdsOdin, 2021-10-01:
Any chance to configure the clients to use insecure connections? I couldn't find any documentation on this.

Leffe64, 2021-10-01:
I got mine working by adding my local network to 'allowed without auth' as per: https://support.plex.tv/articles/200890058-authentication-for-local-network-access/
Then on your Rasplex, logout from your Plex-account and add your server IP in Preferences/Network/Media Server/Manually specify server.

sdsOdin, 2021-10-01:
Thank you @Leffe64 I will give that a try!

bcb2000, 2021-10-01:
Above workaround would not be great for me, because I have individual users set up in my plex account.  As soon as I log out my RasPlex, I lose my multi-user setup (please correct me if I'm wrong here)

Tell me if I'm asking a dumb question here... but shouldn't it be possible to add some more recent root certs (including whichever it is that's needed from this page https://letsencrypt.org/certificates/) into the RasPlex image?

dalehamel, 2021-10-01:
> Tell me if I'm asking a dumb question here... but shouldn't it be possible to add some more recent root certs (including whichever it is that's needed from this page https://letsencrypt.org/certificates/) into the RasPlex image?

Yes, I had this idea last night.

The image is squashfs. This could be achieved with:

- Unqsuashfs'ing the image (extracting it)
- Updating the ca-certificates directory with the most recent ca certs
- Re-squashfs'ing the image

This would then require re-flashing the new image.

I believe that this could also be tested out without reflashing to confirm that updating the CA certs fixes the issue, by mounting a tmpfs folder over-top of the ca-cert directory, then extracting the new CA certs into the tmpfs folder. This could also be a temporary, stop-gap solution.

I'm a bit busy today but if I have time, i'll try one or both of these.

dalehamel, 2021-10-01:
I believe this is the bundle that we need to update to:

https://raw.githubusercontent.com/LibreELEC/LibreELEC.tv/master/packages/security/openssl/cert/cacert.pem

And it needs to overwrite `etc/ssl/cert.pem`

dalehamel, 2021-10-01:
Here is the process I followed:

```
mkdir img sfs
sudo mount -o loop,offset=4194304 RasPlex-1.8.0.148-573b6d73-RPi2.arm.img img
cp img/SYSTEM sfs
cd sfs
unsquashfs SYSTEM
cd squashfs-root
wget https://raw.githubusercontent.com/LibreELEC/LibreELEC.tv/master/packages/security/openssl/cert/cacert.pem
mv cacert.pem etc/ssl/cert.pem
cd ..
rm SYSTEM
mksquashfs squashfs-root SYSTEM -comp gzip
cd ..
cp sfs/SYSTEM img/SYSTEM
md5sum img/SYSTEM # update the img/SYSTEM.md5 file with the new checksum
sudo umount img

```

Here is the image to try out https://github.com/RasPlex/RasPlex/releases/download/1.8.0b/RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz

Leffe64, 2021-10-01:
Since I've got a 3+ I can't try your image but I tried resetting my changes and mounted a tmpfs over the /etc/ssl, replaced the cert.pem with the new ca-cert's but to no avail :( Strange. I thought it was a good idea. :)

CooLeRRRS, 2021-10-01:
> Since I've got a 3+ I can't try your image but I tried resetting my changes and mounted a tmpfs over the /etc/ssl, replaced the cert.pem with the new ca-cert's but to no avail :( Strange. I thought it was a good idea. :)

The address of the certificates is slightly different, like /etc/pki/ssl/tls/cacert.pem

dalehamel, 2021-10-01:
> The address of the certificates is slightly different, like /etc/pki/ssl/tls/cacert.pem

That should just be a symlink to the file in /etc/ssl

```
ls -l squashfs-root/etc/pki/tls/certs 
total 0
lrwxrwxrwx 1 dale.hamel dale.hamel 17 Mar 16  2017 ca-bundle.crt -> /etc/ssl/cert.pem
```

dalehamel, 2021-10-01:
I was able to do my own test, and yeah unfortunately it also seems to not be working, so perhaps the ca certs are not the only issue here

CooLeRRRS, 2021-10-01:
The problem is not with the player, but with the server. Since yesterday everyone has been unable to access it from outside.

dalehamel, 2021-10-01:
> The problem is not with the player, but with the server. Since yesterday everyone has been unable to access it from outside.

I thought it might be both, since the CA cert bundle included with rasplex is fairly old, it stands to reason there could have been SSL errors due to the root trust cert expiring.

I cannot connect my local rasplex instance to even my local server. I am able to sign in to plex using the /link endpoint (which is odd if there was an SSL issue, i would expect no connections to work), but no media servers show up at all.

I also tried https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932099087 but doesn't seem to be working in my case

CooLeRRRS, 2021-10-01:
I disabled all certificates on the server. Connected locally, authorized the Malinka and everything worked inside the network.

dalehamel, 2021-10-01:
Yes, @CooLeRRRS is correct about:

> The problem is not with the player, but with the server. Since yesterday everyone has been unable to access it from outside.

I was able to get the older image to connect to plex by doing the following:

- In the plex server settings, adding the LAN ip addresses as allowed without auth https://support.plex.tv/articles/200890058-authentication-for-local-network-access/
- In the plex server settings, changing the "secure connections" field from "preferred" to "disabled"
- Logging out of plex on Rasplex
- directly specifying the server address in rasplex.

Note this is the ORIGINAL image, not the one with the updated root certs.

So this is absolutely definitively NOT anything to do with the CA cert bundle on rasplex itself, this is definitely an issue with the infrastructure managed by plexinc.

Leffe64, 2021-10-01:
If you change the Secure setting to 'Disabled' you dont need to change anything else. At least this works for me. But of course 'Prefered' would be prefered :)

Leffe64, 2021-10-01:
But then again.. It stopped working, again. :( Will be using my original settings which still seems to be ok.

alcroito, 2021-10-01:
I'm using OpenPHT on macOS, which is apparently a fork of RasPlex.

Replacing the cacert.pem file from https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932194913
in  /Applications/OpenPHT.app/Contents/Resources/XBMC/system/cacert.pem

did not help.

But replacing it with the one linked at https://www.reddit.com/r/usenet/comments/pyo27t/ndngd_lets_encrypt_root_certificate_expiration/hevvwol/
helped.

Direct link: https://nzbget.net/info/cacert.pem

Maybe it helps for others with RasPlex as well.

christopher-daly, 2021-10-01:
> I'm using OpenPHT on macOS, which is apparently a fork of RasPlex.
> 
> Replacing the cacert.pem file from [#645 (comment)](https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932194913) in /Applications/OpenPHT.app/Contents/Resources/XBMC/system/cacert.pem
> 
> did not help.
> 
> But replacing it with the one linked at https://www.reddit.com/r/usenet/comments/pyo27t/ndngd_lets_encrypt_root_certificate_expiration/hevvwol/ helped.
> 
> Direct link: https://nzbget.net/info/cacert.pem
> 
> Maybe it helps for others with RasPlex as well.

I gave this a shot and still seeing the error in logs.
```ERROR: CCurlFile::FillBuffer - Failed: Peer certificate cannot be authenticated with given CA certificates(60)```

dalehamel, 2021-10-01:
I also tried updating the cert, and it looks like that is not sufficient:

https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

>  if clients of your API are using OpenSSL, they must use version 1.1.0 or later.

I believe this is the (new) issue, as rasplex is on openssl-1.0.2o

https://github.com/RasPlex/OpenPHT-Embedded/blob/244a5af9eea3d54c79153f415619259d65de5418/packages/security/openssl/package.mk

I'll see about bumping openssl as well as the cert.

alcroito, 2021-10-01:
The macOS OpenPHT is using 1.0.2k.
https://github.com/RasPlex/OpenPHT/blob/cf23f65fd009a163c417239a26be3cc6060c9882/tools/darwin/depends/openssl/Makefile#L7

I flipped around which cacerts.pem my OpenPHT should use, was checking the logs, and I definitely saw a difference that with the new certs i didn't get 
`ERROR: CCurlFile::FillBuffer - Failed: Peer certificate cannot be authenticated with given CA certificates(60)`
anymore. Same with the server logs.
So not sure why would there be a difference for regular RasPlex. I thought I'd at least share the data point.

dalehamel, 2021-10-01:
@alcroito yeah so i found https://www.openssl.org/blog/blog/2021/09/13/LetsEncryptRootCertExpire/ which seems to indicate the issue is actually the **presence** of the expired cert (DST Root CA X3), which i guess is also used to sign the new cert (ISRG Root X1)

They give a work-around which looks to be basically just deleting that cert, so i'm giving this a shot.

dalehamel, 2021-10-01:
Yes, can **confirm** that fixes the issue. 

I have removed the DST Root CA X 3 certificate, and now when I flashed rasplex it is able to connect just fine, with our without signing in to plex.

This is definitely a client-side issue **WITH RASPLEX**, not plex's fault (i retract my previous statement to that effect).

The new certificate is in fact present in rasplex, the problem is that it is signed with this expired certificate AND on an older openssl which doesn't fall back to just using the new certificate. 

I've uploaded the confirmed working image to https://github.com/RasPlex/RasPlex/releases/download/1.8.0b/RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz for others to try.

I'll re-pack the RPI 1 image and the 3b+ prerelease image as well.

dalehamel, 2021-10-01:
This is the confirmed working pem file, which is just the latest pemfile generated by mk-ca-bundle.pl, ensuring the offending expired cert is removed.

[cacert.pem.txt](https://github.com/RasPlex/RasPlex/files/7270212/cacert.pem.txt)

ark626, 2021-10-01:
I solved this issue with the following way:

Download this cert Direct link: https://nzbget.net/info/cacert.pem

Create a copy of the ssl folder
`cp -a /etc/ssl/ /storage/`

Mount the folders to link them
`mount -o bind /storage/ssl/ /etc/ssl/`

go to the /storage/ssl folder and delete the cacert.pem file.
Insert the cacert.pem from the directlink

```
cd /storage/ssl
rm cacert.pem
wget https://nzbget.net/info/cacert.pem
```
Make the link of the folders as autostart:

`echo "mount | grep -q etc/ssl || mount -o bind /storage/ssl/ /etc/ssl/" >/storage/.config/autostart.sh`

Now go to the plex server and fix the cached cert:

    Browse to the Plex Media Server data directory on your server (https://support.plex.tv/articles/202915258-where-is-the-plex-media-server-data-directory-located/)
    Open the Cache folder
    Delete cert-v2.p12
    Restart Plex Media Server
    
   Restart rasplex

dalehamel, 2021-10-01:
The solution by @ark626 should work for fixing existing installations in-situ. I'll upload fixed images with the correct certificate.

ark626, 2021-10-01:
> 
> 
> The solution by @ark626 should work for fixing existing installations in-situ. I'll upload fixed images with the correct certificate.

I would appreciate a new image very much since rasplex seems to be the only installation that runs without further issues or performance drops on raspberry pi 3

caring4apathy, 2021-10-01:
Thank you! Thank you! Thank you!

Just tried that new image you posted 15 mins ago with a fresh install, and it works. You've just saved me a lot of trouble and am greatly appreciative. Had to create a Github account just to say thanks :)

dalehamel, 2021-10-01:
I have repacked all supported images at https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b including the 3b+ prerelease image.

I would like to call out @ark626 's elegant solution as one that should work for all existing installs without reflashing, thanks again for that!

I really appreciate EVERYONE who has helped to debug this, and especially @alcroito whose comment led to the solution here.

dalehamel, 2021-10-01:
I am going to leave the issue open for the time being for the sake of visibility.

**Please only comment on the issue if you have problems and none of the solutions posted here are working for you**

The solutions, in order of preference are:

1. Re-flash with the updated image with the updated CA bundle
2. Use @ark626 's workaround to update existing installations
3. Disable "secure connections" on the plex server, per https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932298511 (last resort, but does work)

sdmfdeeger, 2021-10-02:
I have no idea how to SSL into the openPHT I have running on an odroid C2 and on an android box I have. There are images for those versions (my android one is the OpenPHT-Embedded-1.8.0.148-573b6d73-Generic.x86_64.img.gz. I have NO idea how to modify these existing images. I don't know how time consuming it is...is there some sort of tutorial for dummies like me, or perhaps someone willing to create images for me. It would be much appreciated. I use OpenPHT everyday on these two devices and it's really disappointing that it doesn't work anymore :(

ark626, 2021-10-02:
> 
> 
> I have no idea how to SSH into the openPHT I have running on an odroid C2 and on an android box I have. There are images for those versions (my android one is the OpenPHT-Embedded-1.8.0.148-573b6d73-Generic.x86_64.img.gz. I have NO idea how to modify these existing images. I don't know how time consuming it is...is there some sort of tutorial for dummies like me, or perhaps someone willing to create images for me. It would be much appreciated. I use OpenPHT everyday on these two devices and it's really disappointing that it doesn't work anymore :(

Maybe this issue in OpenPHT helps https://github.com/RasPlex/OpenPHT/issues/325#issue-1013772253

Generally you download a SSH Client like putty and connect via IP user root password plex (at least for rasplex for OpenPHT i dont know the user) afterwards you move into the directory where the cacert.pem is stored.

Delete it and replace it with this one https://nzbget.net/info/cacert.pem.
(wget https://nzbget.net/info/cacert.pem)

Then reboot the device. If the distribution allows to delete the file only of course.
If it is similar to rasplex (libreelec ) then you sadly cannot modify the files since it a read only file.

If this is the case somebody has to compile an image that you have to flash.

bcb2000, 2021-10-02:
Thank you all for your work on this.

I didn't want to lose my settings, but I couldn't make @ark626 's solution work for me, so instead I downloaded the new image to my RasPlex device, extracted the SYSTEM and SYSTEM.md5 files from the image and copied them into the /flash partition on my RasPlex devices.  

```
# get new image (RasPlex for Pi 2 for me)
wget https://github.com/RasPlex/RasPlex/releases/download/1.8.0b/RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz
# decompress
gunzip RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz

# loop device for first partition, and mount
losetup -o 4194304 /dev/loop1 RasPlex-1.8.0.148-573b6d73-RPi2.arm.img
mkdir mnt
mount -t vfat /dev/loop1 mnt/

# remount /flash writable
mount -o remount,rw /flash
 
# backup existing SYSTEM files
mv /flash/SYSTEM /flash/SYSTEM-20170317
mv /flash/SYSTEM.md5 /flash/SYSTEM-20170317.md5

# copy over new SYSTEM files
mv mnt/SYSTEM* /flash/

# checksums
md5sum /flash/SYSTEM
cat /flash/SYSTEM.md5

# remount /flash read-only
mount -o remount,ro /flash

# tidy-up
umount mnt
rm RasPlex-1.8.0.148-573b6d73-RPi2.arm.img

# reboot
reboot

```
All is well in the world again.  Thanks again.

ark626, 2021-10-02:
> 
> 
> Thank you all for your work on this.
> 
> I didn't want to lose my settings, but I couldn't make @ark626 's solution work for me, so instead I downloaded the new image to my RasPlex device, extracted the SYSTEM and SYSTEM.md5 files from the image and copied them into the /flash partition on my RasPlex devices.
> 
> ```
> # get new image (RasPlex for Pi 2 for me)
> wget https://github.com/RasPlex/RasPlex/releases/download/1.8.0b/RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz
> # decompress
> gunzip RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz
> 
> # loop device for first partition, and mount
> losetup -o 4194304 /dev/loop1 RasPlex-1.8.0.148-573b6d73-RPi2.arm.img
> mkdir mnt
> mount -t vfat /dev/loop1 mnt/
> 
> # remount /flash writable
> mount -o remount,rw /flash
>  
> # backup existing SYSTEM files
> mv /flash/SYSTEM /flash/SYSTEM-20170317
> mv /flash/SYSTEM.md5 /flash/SYSTEM-20170317.md5
> 
> # copy over new SYSTEM files
> cp mnt/SYSTEM* /flash/
> 
> # checksums
> md5sum /flash/SYSTEM
> cat /flash/SYSTEM.md5
> 
> # remount /flash read-only
> mount -o remount,ro /flash
> 
> # tidy-up
> umount mnt
> rm RasPlex-1.8.0.148-573b6d73-RPi2.arm.img
> 
> # reboot
> reboot
> ```
> 
> All is well in the world again. Thanks again.

Sorry for that. I also recognized it. It seems that the certificate is loaded faster than the autoload I configured or something else.
But tbh. this is above my knowledge so i attached your solution into my comment.

aniel300, 2021-10-02:
what about for those that uses openpht like me ?

g1t-dlanor, 2021-10-02:
> 
> 
> what about for those that uses openpht like me ?

For OpenPHT running under a Windows OS you can use the "cacert.pem" posted by Dale a few posts earlier.
That file should be stored into the subfolder named "system" inside the folder where "OpenPHT.exe" resides.

As for OpenPHT on other platforms I don't know the details, but I assume the certificate would be in a location corresponding to what has been described for RasPlex in posts above, and replacing it will most likely require remounting that filesystem as writable.

closed 2021-10-02 by g1t-dlanor

aniel300, 2021-10-02:
ahh gotcha, i will give it a try later on, thanks

caring4apathy, 2021-10-02:
Am I completely crazy, or is the default ftp/ssh login and pass not working on this new image? I use the Black Edition skin, which I thought came by default with Rasplex as well, but went to manually install it and I can't ftp or ssh into my Rasplex with the new image using root/rasplex login.

caring4apathy, 2021-10-02:
@dalehamel Myself and another person on Reddit using this new image can't ssh or ftp into the new image using the default root/rasplex login. Any idea why this may be?

fwicht, 2021-10-02:
> @dalehamel Myself and another person on Reddit using this new image can't ssh or ftp into the new image using the default root/rasplex login. Any idea why this may be?

Yep, same here cannot ssh with default user/password (root/rasplex).

dalehamel, 2021-10-02:
@caring4apathy yep that's my bad, I made a mistake repacking the images. I (accidentally) didn't use root when i re-ran `mksquashfs` and it ommitted the shadow file needed for passwords. I'm re-repacking the images now, and i'll re-upload them once I've confirmed that I've fixed the issue.

In my haste to fix the one issue, I made another! Woops! Sorry folks. As a workaround, ssh key login **should** still work on the affected images.

Syco54645, 2021-10-02:
> I have repacked all supported images at https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b including the 3b+ prerelease image.
>
> I would like to call out @ark626 's elegant solution as one that [...]

Any chance we can get x86 images? I am running this on a NUC and am unable to fix the issue using the workaround that ark626 shared above.

dalehamel, 2021-10-02:
Since people keep asking, i made a script to do this for OpenPHT.

https://gist.github.com/dalehamel/369a6670102be1db02b13df539a7c8c6

**Please stop posting about OpenPHT on this issue. If you need to post, post on https://github.com/RasPlex/RasPlex/issues/646**

Leffe64, 2021-10-03:
@dalehamel, @ark626 : Thanks for all your efforts, everything works like a charm.

jishi, 2021-10-03:
Will this allow the in app update flow? I only have a remote RasPlex at a friend that I can't access physically atm, so wondering if they can just select update.

Risca, 2021-10-03:
I just tried it, and unfortunately no. I believe an update needs to be packaged for normal update flow to work. For now, fresh reinstall is the easiest solution.

dalehamel, 2021-10-03:
@jishi @Risca 

Unfortunately the remote update process no longer works. You will need to have your friend either:

1. Re-flash rasplex at their computer
2. Copy the SYSTEM and SYSTEM.md5 files manually to their SD card at a computer
3. Copy the SYSTEM and SYSTEM.md5 files over SSH

bArAkA01, 2021-10-06:
> Thank you all for your work on this.
>
> I didn't want to lose my settings, but I couldn't make @ark626 's solution work for me, so instead I downloaded the new image to my RasPlex device, extracted [...]

Hello, I just did that on my Pi 3b+ : 
now it doesn't boot properly as before , I have this message 
NEVER TOUCH boot= in extlinux.conf / cmdline.txt!
If you don't know what you are doing,
your installation is now broken.

Normal startup in 60s...

then it counts down to 0 and boots properly

-> not unworkable but kind of a pita

But, main issue is not having Bluetooth working anymore.
In Service Bluetooth is enabled, but in Bluetooth it says Bluetooth is disabled, so I cannot pair the remote control.

Any suggestions ?

regards,
Julien

bArAkA01, 2021-10-07:
> Hello, I just did that on my Pi 3b+ : now it doesn't boot properly as before , I have this message NEVER TOUCH boot= in extlinux.conf / cmdline.txt! If you don't know what you are doing, your [...]

Guess I took the wrong image, had to be RasPlex-1.8.0b.148-573b6d73-RPi2.arm.img.gz cause my device is a 3b not plus.
testing right now, reverting soon

jwatilo, 2021-10-10:
Just a word of THANKS to @dalehamel  and all of the others who diagnosed the issue and built new RasPlex images. I had my RPi2 back up and running in no time thanks to YOU!!!!!!!!!!!!