RasPlex/RasPlex #646: Can we get x86 Generic Images?

Syco54645, 2021-10-02:
I am running OpenPHT-Embedded-1.8.0.148-573b6d73-Generic.x86_64.img.gz on my Intel Nuc and am unable to resolve the ssl error. Is there any way that we can get x86 generic images?

dalehamel, 2021-10-02:
Here is a script that should be able to do it for you https://gist.github.com/dalehamel/369a6670102be1db02b13df539a7c8c6

If i have time I may use it to repack the images and make a new release later.

Syco54645, 2021-10-02:
> Here is a script that should be able to do it for you https://gist.github.com/dalehamel/369a6670102be1db02b13df539a7c8c6
> 
> If i have time I may use it to repack the images and make a new release later.

Thank you. Is there anything special to running this? The only output I get is "Checking required dependencies..." and then the script appears to exit.

dalehamel, 2021-10-02:
@Syco54645 run the script with `bash -x ./repack.sh` and it will show which dependency is missing, my bad.

Or, just wait a minute, because I'm going to upload all of the embedded images repacked shortly.

Syco54645, 2021-10-02:
> @Syco54645 run the script with `bash -x ./repack.sh` and it will show which dependency is missing, my bad.
> 
> Or, just wait a minute, because I'm going to upload all of the embedded images repacked shortly.

I will wait because it is having a few different errors, not just permission related. I will test when I see the repacked images hit.

dalehamel, 2021-10-02:
https://github.com/RasPlex/OpenPHT/releases/tag/v1.8.0b should have all of the embedded images repacked.

I'm leaving it as a "prerelease" until someone can confirm these images are working for them.

dalehamel, 2021-10-02:
@aniel300 @sdmfdeeger since you asked on the RasPlex issue

Syco54645, 2021-10-02:
> https://github.com/RasPlex/OpenPHT/releases/tag/v1.8.0b should have all of the embedded images repacked.
> 
> I'm leaving it as a "prerelease" until someone can confirm these images are working for them.

I can confirm generic is working on my nuc6cay. I cant remember if this can take the EGL build or not but am going to try it and report back.

Syco54645, 2021-10-02:
@dalehamel both generic and the egl image work fine on my nuc.

dalehamel, 2021-10-02:
Since it was the same process for all images, i'll assume they are all fixed. I've marked the repacked release as a full release, not just a prerelease.

sdmfdeeger, 2021-10-02:
Any chance of getting the s905 image on this page repacked like this as well? I just realized my android box is using this image and not the generic one. https://github.com/Veisen/OpenPHT/releases

dalehamel, 2021-10-02:
@sdmfdeeger no idea whose fork that is, but in theory the same repacking script should work on them. Since they aren't our images, there isn't any practical way for us to re-host them. Give https://gist.github.com/dalehamel/369a6670102be1db02b13df539a7c8c6 a try?

sdmfdeeger, 2021-10-02:
dalehamel, is there a certain program you are using to run that script? I'm sorry, I am just ignorant to this whole process. Don't mind doing it myself, just unsure how to do it.

RolfHauck, 2021-10-31:
Hi, Dale, is your new version 1.8.0.b supposed to run with any Plex server version, or do I have to update the server also? My local Plex server is running 1.20.3.3437. And I still don't get a connection to my server, but the following log messages on the server:
[CERT/OCSP] HTTP 200 response from GET http://r3.o.lencr.org/
ERROR - [CERT/OCSP] response error: unauthorized.
INFO - [CERT/OCSP] couldn't fetch a valid response; retrying in 10800 seconds