RasPlex/RasPlex #647: 1.8.0b: System Settings not accessible

muadib25, 2021-10-03:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
System settings option is not accessible (a message "Settings addon is not yet ready, please try again later" comes up), and login is not saved after reboot.
This is with a fresh reflashing from 1.8.0b image.
Also, Automatic Sign-in via Prefs -> Settings -> Plex is not selectable
[Always attempt to reproduce your issue in the default skin]
Skin: OpenPHT (only entry)
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: RPi 3B+
**RasPlex**: 1.8.0b
**PMS**: RasPlex 1.8.0.0 573b6d73
**Network**: ethernet

dalehamel, 2021-10-03:
Which RasPlex image is this of the three from https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b ?

muadib25, 2021-10-03:
It seems that my MicroSD gave up working because when I tried to reflash it wouldn't complete the process.
So I tried with a new one which was successful and none of the above issues were repeated.

closed 2021-10-03 by muadib25

Eternal-plasma, 2021-10-03:
try using a surveillance grade sd card for longer life worked for me :)

muadib25, 2022-05-14:
> 

Thanks, I'll try one in the future! :)