RasPlex/RasPlex #648: Remote server can’t use

kkishnani, 2021-10-03:
Remote server on rasplex was working fine until it stopped for some reason… 
It works on iOS devices nicely just doesn’t work on rasplex. 
Plex server website also says connected with outside network. 

**RPi model**: [3B+]
**RasPlex**: [1.8.0 default skin]
**PMS**: [1.24.3.5033 and synology dsm7.0]
**Network**: [ethernet]

nightmare04, 2021-10-03:
Yes, remote server not work with rasplex client

kkishnani, 2021-10-03:
> Yes, remote server not work with rasplex client

But It did work for couple of days at first.

nightmare04, 2021-10-03:
> > Yes, remote server not work with rasplex client
> 
> But It did work for couple of days at first.

Yes

kkishnani, 2021-10-03:
> > > Yes, remote server not work with rasplex client
> > 
> > 
> > But It did work for couple of days at first.
> 
> Yes

So what do you mean? It works for first few days and then it stops? I don’t think so it should either work or it should not…

nightmare04, 2021-10-03:
It works many years, and stop work yesterday 1 October

dalehamel, 2021-10-03:
This is a duplicate of https://github.com/RasPlex/RasPlex/issues/645

The problem is that on Sept 30, the root CA certificate that LetsEncrypt service uses (which Plex Inc uses) expired. There is a bug in the old version of OpenSSL that Rasplex uses where if this expired root CA exists, it fails to fallback to the new root CA, even though it has it.

We fixed this issue by repackaging the images to update the CA certificates https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b

Update rasplex by either re-flashing the image, or copying the SYSTEM and SYSTEM.md5 file to your SD card.

closed 2021-10-03 by dalehamel