RasPlex/RasPlex #649: 1.8.0b | ssh default root password

whoo, 2021-10-03:
Hi Guys,

root/rasplexe password didn't work after fresh install.
I saw that <code>/etc/shadow</code> is 0k size problem on install process ?

Thanks for your help.

dalehamel, 2021-10-03:
@whoo when I first created the release on Oct 1, I made a mistake as this comment says https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932756892 and didn't use root when runnnig `mksquashfs` to repack the images. 

I fixed the issue yesterday, the images at https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b should work. Just reinstall these images.

If that is still not working, please give the filename and shasum of the image you are tyring that is not working, and reopen this issue.

closed 2021-10-03 by dalehamel

whoo, 2021-10-03:
Hi,
I'll try.
Thanks.