RasPlex/RasPlex #650: remote server not working

nightmare04, 2021-10-03:
Remote server stop working 1 october, reinstall image with version 1.8.0B, still not working.
RPi model: RPi B+
RasPlex: 1.8.0b
PMS: RasPlex 1.8.0.0 573b6d73
Network: ethernet
[image]
[image]

dalehamel, 2021-10-03:
Have you signed in?

nightmare04, 2021-10-03:
> Have you signed in?

Yes, try add local server manual - not working

dalehamel, 2021-10-03:
If you are on the new images (https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b) it is probably not an SSL issue like in #645 , but try setting "Secure connections" to "disabled" on the server like in https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932298511

I can't really be much help beyond this though

nightmare04, 2021-10-03:
> If you are on the new images (https://github.com/RasPlex/RasPlex/releases/tag/1.8.0b) it is probably not an SSL issue like in #645 , but try setting "Secure connections" to "disabled" on the server like in [#645 (comment)](https://github.com/RasPlex/RasPlex/issues/645#issuecomment-932298511)
> 
> I can't really be much help beyond this though

Thnx, with "Secure connections" to "disabled" work

closed 2021-10-03 by nightmare04

dalehamel, 2021-10-03:
That's very weird, the updated SSL certificate authority should make that not be necessary. Perhaps it is a server-side issue

ark626, 2021-10-03:
YOu have to delete the cached certificate in the Server.
Now go to the plex server and fix the cached cert:

```
Browse to the Plex Media Server data directory on your server (https://support.plex.tv/articles/202915258-where-is-the-plex-media-server-data-directory-located/)
Open the Cache folder
Delete cert-v2.p12
Restart Plex Media Server
```