RasPlex/RasPlex #655: Bootloop at user selection screen

jrizz43, 2022-03-11:
I was able to successfully connect to my plex server over wifi and log in with my default plex user. When i go to switch managed user accounts it goes to the "Choose User" screen. When I click on ANY of the available users it reboots the Pi and cycles back to that user selection screen with no option to choose any other settings. Screenshot available if needed.

[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: [raspberry pi model] Pi Model B
**RasPlex**: [version and skin] Unable to tell/ Default skin
**PMS**: [version and operating system] Windows 10 4.76.1
**Network**: [ethernet/wifi] Wifi

JonasSvegland, 2022-07-28:
I have the same issues. Everything worked fine, but once I tried to select another user Rasplex crashed and rebooted. Since then I'm stuck in user selection screen, and Rasplex reboots what ever user I select.

Image RasPlex-1.8.0.0b-573b6d73-RPi3bplus.arm.img.gz

RPi model: Raspberry Pi 3 Model B+
RasPlex: [version and skin]  Default skin
Network: [ethernet/wifi] Ethernet

Makhuta, 2023-06-09:
I've had the same issue.
It has to be something with the passwords being blank or something, if I put my password for desired profile (or anything if there isn't any password nothing will show up) with keyboard attached to Pi and then press Enter it will load up normally. That is the only way I currently found.