RasPlex/RasPlex #656: Can’t access server externally

Cathaldub, 2022-05-05:
Having problem where I can’t see my PleX library when I’m outside of my local network, it sounds like the ssl issue in #645 but I’m using the latest version linked in those threads, I’ve tried re downloading, but always the same, it’s a pi 3b+
I’ve cleared the cached cert of my server which is running the latest Linux version of PleX server.

using kodi and PleX plug in works fine but I’d prefer use rasplex, anyone any ideas?

Spark-NV, 2022-05-07:
Without logs no one will be able to help very much.

But for testing if its indeed the cert if you want you can try my fork of openpht-embedded as it has the latest cert installed but i  never compiled a rasplex img. so it will be labeled as openpht, but its the same thing as rasplex without the rasplex loading screen essentially.

Cathaldub, 2022-05-07:
Thanks Spark-NV, I’ll give that build a try, I wasn’t sure where to find the logs on rasplex, there was nothing of interest on the server side.